google/xls #262: Add formatter/linter config files

meheff, 2021-01-19:
To make it easy to run the linters and formatters for C++ (clang-tidy, clangformat, cpplint) and python (pylint, yapf) we should add config files for these various utilities and add instructions for how to run them. They should be configured to match internal Google style to minimize impedance mismatch.

cdleary, 2021-01-20:
Related: https://github.com/google/xls/issues/125

meheff, 2023-03-04:
@hzeller Is this done now?

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): Repository-level formatter and linter configuration files have been added at the root of the open-source repository, including `.clang-format`, `.clang-tidy`, `.pylintrc`, `.editorconfig`, and `.pre-commit-config.yaml`. (Closed by Gemini)

closed 2026-11-14 by allight