google/xls #288: Multi-port write register-file-like structure characterization

cdleary, 2021-02-01:
This issue tracks a client implementation (of the synthesis server) that checks the timing closure for a register-file-like structure that has N write ports. In general the sweep space is N entries, B bits wide, W write ports, and I believe we currently want to assume we can take all of the values out (broadside) on a given clock. Initial values are something like N=80 B=22 W=4 but we can scale that down for unit testing purposes.

RobSpringer, 2021-02-25:
Ah! This slipped under my radar. This exists - it just needs to be exported to OSS-land. I'll do that move Real Soon Now.

cdleary, 2022-11-02:
CC @grebe could be useful to compare the characterization of this RTL artifact to RAM support for a given platform and see where in terms of elements/width the area/power curves flip from preferring one solution vs the other.

(I forget what portedness options open ram generation solutions for sky130 support, I guess even if it's just single ported and you have BYO multi port solution on top it'd still be interesting to know how the single port curves compare.)

grebe, 2022-11-03:
Ah, interesting. Agree it seems useful to be able to sweep PPA for a computation over a suite of memory-ish constructs. I think OpenRAM has 1RW and 1R1W variants, as well as flop arrays, so I think there could be an interesting comparison.

allight, 2026-11-14:
Closing this issue as obsolete (accepted by @allight): The standalone register-file characterization script and `SynthesisService` regfile sweep discussed in this issue were superseded by the general operation characterization client (`xls/synthesis/op_characterization_client.py` and `xls/synthesis/op_characterization_client_main.py`) and RAM rewrite/codegen support (`xls/passes/ram_rewrite_pass.h`). (Closed by Gemini)

closed 2026-11-14 by allight