google/xls #290: Add fail_unless! and fail_when! to DSLX

meheff, 2021-02-03:
These expressions take a condition and fail! when the condition is true/false.

cdleary, 2021-02-18:
Note that this one is blocked on #232 -- once we have the ability to handle "unconditional fails" it should be comparatively easy to add another conditional into the mix.

cdleary, 2021-02-19:
Or, it's not necessarily blocked, just if we add these before we introduce the changes for #232 these will be affected so will need to be updated as well.

closed 2024-03-15 by copybara-service[bot]

cdleary, 2025-01-14:
Note that what landed was `assert!(predicate, label)` in https://github.com/google/xls/commit/08378787172a10d97516421237a508641bec3d00 as documented by https://google.github.io/xls/dslx_std/#fail-assert-assertion-failure