google/xls #291: DSLX dynamic width-slice referential transparency violation in interpreter impl

cdleary, 2021-02-07:
Something I stumbled across is we currently pass type checking but fail in the interpreter for an expression like the following:

```dslx
let x: u7 = ...;
let i = u26:-1;
let sliced: u32 = x[i+:u2];
```

```dslx
let x: u7 = ...;
let sliced: u32 = x[u26:-1+:u2];
```

The interpreter I believe does not blow up on the former but blows up on the latter, because its implementation is taking a different path when the start is a literal number, and you get something like "u26:-1 does not fit in 7 bits". I think this cropped up because we wanted to let people avoid annotating the width of the start and infer it from the subject being sliced.

juttadegener, 2023-06-09:
Dibs.

proppy, 2026-10-28:
I don't think this apply to tiv2 anymore (I also don't think we support negative width slice) /cc @richmckeever

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `PopulateTableVisitor::HandleWidthSliceInternal` (`xls/dslx/type_system_v2/populate_table_visitor.cc:1396`), `BytecodeEmitter::HandleWidthSlice` (`xls/dslx/bytecode/bytecode_emitter.cc:1148`), and `BytecodeInterpreter` (`xls/dslx/bytecode/bytecode_interpreter.cc`) now enforce unsigned dynamic width-slice start indices and evaluate inline and variable start expressions identically. (Closed by Gemini)

closed 2026-11-14 by allight