google/xls #300: Fuzzer not generating a lot of opcodes

meheff, 2021-02-11:
No shifts, selects, counted fors, and other ops. Probably fall out from the port to C++.

cdleary, 2021-02-12:
My bad for the porting bug, I should have thought to look at the summary per https://google.github.io/xls/fuzzer/#summaries !

cdleary, 2021-02-12:
The one value of the golden reference files is they let you eyeball what kind of computation comes out of the fuzzer with any modification -- I'm just not sure whether implementing the record and replay layer on top of the RNG is worth it. Maybe could have caught this if we had a golden reference to look at... or maybe we just try to be vigilant about running the fuzzer summary on any fuzzer change. WDYT?

meheff, 2021-02-12:
I don't know if the golden reference would have necessarily caught it. It'd be very unlikely that you'd notice that shifts weren't being generated, or that 90% of the return values were parameters rather than 10%.  A better test might be to run the fuzzer for a little while (1000 samples?) and verify all the ops show up in the summary. And maybe also check that the vast majority of the optimized samples aren't trivial (single node) which was the case when most samples were dead code.

meheff, 2021-02-18:
The following commit fixed the issues caused by the port of the fuzzer to c++ (eg, no shifts being generated):

https://github.com/google/xls/commit/a103498fd1861a326ad5f08f2b8335df12c6b33f

However, we're still thin on array ops. We only generate Op::kArray operations with Literal operands and create no other array operations. We should generate more insteresting  kArray ops as well as array-typed return values and parameters, array index, array update, and multi dimensional array operations.

cdleary, 2021-03-08:
I think this commit may have addressed the last comment in full? https://github.com/google/xls/commit/be383902dc046ffa559abb6efae2ac10d06ed9d9

meheff, 2021-03-08:
One last thing to do might be to verify that most of the optimized samples are not trivial. At least twice in the past we've had issues were the majority of optimized samples were a single node or just constants, and this is an easy to miss bug. Maybe check that at least half of optimized samples have five non-literal, non-param nodes.

meheff, 2021-06-07:
Haven't gotten to the last comment suggestion (verify most optimized samples aren't trivial), but I think this issue can be closed as we have tests of op coverage in the fuzzer.

closed 2021-06-07 by meheff