google/xls #304: Parametric Functions: Can't call parametric function to define a paramteric value.

jbaileyhandle, 2021-02-17:
In the following code, the parametric value IDX_BITS is defined by calling the paramteric function std::clog2 :
```
fn dot_product<BITCOUNT: u32, LENGTH: u32, 
  IDX_BITS: u32 = std::clog2<u32:32>(LENGTH)>
  (a: bits[BITCOUNT][LENGTH], b: bits[BITCOUNT][LENGTH]) -> bits[BITCOUNT]{
  
  for(idx, acc): (bits[IDX_BITS], bits[BITCOUNT])
    in range (u32:0 , LENGTH) {
      
    let partial_product = a[idx] * b[idx];
    acc + partial_product
  } (u32:0)
}

fn main(a: u32[4], b: u32[4]) -> u32 {
  let result = dot_product(a, b);
  u32:0
} 
```
This fails to compile from dslx to ir with the following error:
```
F0217 17:13:48.770722       3 ir_converter_main.cc:135] Check failed: ::absl::OkStatus() == (status) (OK vs. INTERNAL: Not enough symbolic bindings to convert function: clog2; need {N} got {}) 
```
Note that we do in fact pass u32:32 to std::clog2 for the parametric value.

If we simply wrap the call to clog2 inside a nonparamteric function as below, we can successfully compile to ir:
```
fn func(a: u32) -> u32 {
  std::clog2<u32:32>(a)
}

fn dot_product<BITCOUNT: u32, LENGTH: u32, 
  IDX_BITS: u32 = func(LENGTH)>
  (a: bits[BITCOUNT][LENGTH], b: bits[BITCOUNT][LENGTH]) -> bits[BITCOUNT]{
  
  for(idx, acc): (bits[IDX_BITS], bits[BITCOUNT])
    in range (u32:0 , LENGTH) {
      
    let partial_product = a[idx] * b[idx];
    acc + partial_product
  } (u32:0)
}

fn main(a: u32[4], b: u32[4]) -> u32 {
  let result = dot_product(a, b);
  u32:0
} 
```

RobSpringer, 2021-02-19:
Looks like this has been fixed recently, possibly by 88a5b05. I'll add a unit test for your case, then close this one.

RobSpringer, 2021-02-19:
I suppose I'd like you to verify once the unit test goes in, just to make sure I've not messed something up in translation, but...I don't think I have? :)

jbaileyhandle, 2021-02-22:
No dice - This issue only comes up if we try to compile to IR, I think.  I still get the same error if I remove "generate_benchmark = False" from the BUILD dslx_test for paramteric_call_paramteric and add this entry function to paramteric_call_parametric.x:
```
fn main(a: u32[4], b: u32[4]) -> u32 {
  dot_product_modref(a,b)
}
```
I've noticed in a few cases that things that work for dslx-only tests fail when we compile to IR - this often seems related to paramteric values.

This isn't a blocking issue for me right now, so please don't let it sidetrack your from more important things.  This probably should be address this at some point, though.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): Calling parametric functions (such as `std::clog2`) inside default parametric binding expressions is supported in DSLX and tested using the exact `dot_product` pattern from the issue in `xls/dslx/tests/parametric_call_parametric.x:23-49`. (Closed by Gemini)

closed 2026-11-14 by allight