google/xls #305: Casting a Const does not preserve Const-ness

jbaileyhandle, 2021-02-17:
Like the title says.  For example, the following code will not compile from dslx to IR because casting Const FOO loses Const-ness (loop range requires constant bounds):
```
const FOO = u8:42;

fn func(a: u32) -> u32 {
  std::clog2<u32:32>(a)
}

fn dot_product<BITCOUNT: u32, LENGTH: u32, 
  IDX_BITS: u32 = func(LENGTH)>
  (a: bits[BITCOUNT][LENGTH], b: bits[BITCOUNT][LENGTH]) -> bits[BITCOUNT]{
  
  for(idx, acc): (bits[IDX_BITS], bits[BITCOUNT])
    in range (u32:0, FOO as u32) {
      
    let partial_product = a[idx] * b[idx];
    acc + partial_product
  } (u32:0)
}

fn main(a: u32[4], b: u32[4]) -> u32 {
  let result = dot_product(a, b);
  u32:0
} 
```
We get the following error when we compile this:
```
F0217 17:34:10.034286       3 ir_converter_main.cc:135] Check failed: ::absl::OkStatus() == (status) (OK vs. INTERNAL: AST node had a non-const IR value: ((FOO) as u32)) 
```

======================================================
As an aside, why can't we just define the Const to be the type we need instead of casting in loop code?  The reason is that I'm actually dealing with parametric values, which are treated as Const values within the function (I think).  See below for desired use case:

```
fn func(a: u32) -> u32 {
  std::clog2<u32:32>(a)
}

fn dot_product<BITCOUNT: u32, LENGTH: u32, 
  IDX_BITS: u32 = func(LENGTH+u32:1)>
  (a: bits[BITCOUNT][LENGTH], b: bits[BITCOUNT][LENGTH]) -> bits[BITCOUNT]{
  
  for(idx, acc): (bits[IDX_BITS], bits[BITCOUNT])
    in range (bits[IDX_BITS]:0 , LENGTH as bits[IDX_BITS]) {
      
    let partial_product = a[idx] * b[idx];
    acc + partial_product
  } (u32:0)
}

fn main(a: u32[4], b: u32[4]) -> u32 {
  let result = dot_product(a, b);
  u32:0
} 
```

======================================================

As a further aside, casting may not be necessary in this or similar cases.  In the following code, we don't explicitly cast the loop bounds to the type of the index, but it still compiles.  Are there cases where casting a paramteric / Const could not be avoided so easily?

```
fn func(a: u32) -> u32 {
  std::clog2<u32:32>(a)
}

fn dot_product<BITCOUNT: u32, LENGTH: u32, 
  IDX_BITS: u32 = func(LENGTH)>
  (a: bits[BITCOUNT][LENGTH], b: bits[BITCOUNT][LENGTH]) -> bits[BITCOUNT]{
  
  for(idx, acc): (bits[IDX_BITS], bits[BITCOUNT])
    in range (u32:0, LENGTH) {
      
    let partial_product = a[idx] * b[idx];
    acc + partial_product
  } (u32:0)
}

fn main(a: u32[4], b: u32[4]) -> u32 {
  let result = dot_product(a, b);
  u32:0
} 
```

cdleary, 2021-02-18:
I think this is related to https://github.com/google/xls/issues/246 (which is somewhere on my plate, but currently active task is plumbing `fail!` to assert IR nodes).

If the constexpr was evaluated at typechecking time and available via the type info it would effectively be a const upper bound (queryable by IR conversion) and this particular use case I think would be resolved.

cdleary, 2021-03-02:
Technically switching to Rust range syntax is probably the most forward-looking way to solve this, since then it's syntactically distinguished from a normal function call -- `range` really want something more like parametric instantiation, i.e. should really be written as `range<START=0, LIMIT=CASTED, LEN=LIMIT-START>() -> T[LEN]`, but because I originally started by making the DSL similar to Python we've temporarily landed here (c.f. https://github.com/google/xls/issues/318#issue-818042528) -- what we can do is change the signature handler for `range` here https://github.com/google/xls/blob/6a132ce9c93ee274f517ff9d948f67df23fa4afb/xls/dslx/dslx_builtins.cc#L241 to force constexpr evaluation of its arguments, which is clearly a superpower only builtins would be able to have, so we can do that and then switch over to range syntax over time as the proper fix.

cc @dkillebrew-g

closed 2021-03-05 by copybara-service[bot]