google/xls #311: dslx_library Bazel build rules

cdleary, 2021-02-25:
We need to figure out how to use bazel depsets to make `dslx_library` and `dslx_test` targets akin to `cc_library` and `cc_test` targets: https://docs.bazel.build/versions/master/skylark/depsets.html We need to "ball up" the transitive dependencies of .x files via this "aspect" that Bazel provides -- right now `.x` files have to be duplicated in all the entry point targets instead.

vincent-mirian-google, 2021-03-27:
@cdleary .

What will be the task of the dslx_library rule? ParseAndTypeCheck?
For the immediate purpose, will dslx_library simply be a placeholder to reference the '.x'?

I assume the dslx_test rule will maintain its role for executing the interpreter, etc...?

Thanks.

cdleary, 2021-03-27:
Right! Rationale being, normally for a `cc_library` rule it'd correspond to a compiling to an object file, whereas for `py_library` and similar there's (nominally) nothing to do except bundle together a bunch of files, we're somewhere in the middle where we don't have an object file to produce but we do want to ensure everything parses and typechecks if you `bazel build` that particular library target.

In the future we can also serialize the type information we compute at the library rule level so it could be loaded by dependent modules instead of needing to be recomputed.

If, down the road, we end up thinking "hey most of our libraries have unit tests, it'd be convenient to ask for both at the same time" we can compose a `dslx_library` and `dslx_test` rule in a convenience/helper genrule, something like `dslx_library_and_test`.

closed 2021-06-19 by vincent-mirian-google