google/xls #316: DSLX: Parametric specification syntax is ambiguous

RobSpringer, 2021-02-26:
Consider the following DSLX:
fn Foo<A: u32> ...

fn Bar() {
  let z = Foo<a>(u32:1);
  ...
}

When parsing, we eventually arrive at Parser::ParseParametrics(), which more-or-less evaluates every comma-separated component of the parametric specification as an Expr.

Parsing this invocation currently fails, as the parser considers "a > u32:1" as the full expression, which is clearly not what's desired here. Unfortunately, cases are not always as clear as this.

What about the invocation
Foo<a > (b) > (c) > (d) ?
This could validly be translated as:
 - ( Foo<a>(b) ) > (c > d)
 - ( Foo<(a > b)>(c) ) > d
 - Foo <a>(b > c > d)
and so on.

Fortunately? Sadly? other languages have a similar ambiguity problem (*why did they all choose < > instead of other delimiters?!?!*), so we can look to them for guidance - in particular, hopefully we can lean on Rust here, since it's influenced so much of our design.

RobSpringer, 2021-02-26:
Sorry - the initial Foo invocation should be:
```
Foo<a > (b) > (c) > (d)
```

cdleary, 2021-02-27:
Notable ref discussed today in solution approach: https://blog.rust-lang.org/2021/02/26/const-generics-mvp-beta.html#no-complex-generic-expressions-in-const-arguments

closed 2021-03-02 by copybara-service[bot]