google/xls #319: Allow arbitrary assert format strings in Verilog

meheff, 2021-03-01:
Often a project may have custom assert macros for use in (System)Verilog. We should support format strings like so:

"`MY_ASSERT({condition}, {message})"

This might result in the following in the generated verilog:

MY_ASSERT(a < 32'h42, "a is too big")

cdleary, 2021-03-08:
I think this may have been resolved by https://github.com/google/xls/commit/24a05b7a957e817f164796293fe0c20f1d87bf27 ?

meheff, 2021-03-08:
There is still connecting it up to the user. What remains to be done is to add an option to codegen_main which specifies the format string.

meheff, 2021-03-08:
One additional thing which might be necessary would be a way to specify tick includes which define the user-defined assert macro. I suppose this could be handled by wrapping it in a genrule to tack on the tick includes at the beginning, but having that is kind of icky.

meheff, 2022-05-12:
This is now supported and wired up to a codegen_main flag.

closed 2022-05-12 by meheff