google/xls #320: Generate packed arrays via VAST codegen

cdleary, 2021-03-01:
Currently we use unpacked arrays, which end up generating lots of unrolled assignments for propagating the array values around. Mark is working on switching us to packed arrays.

I believe https://github.com/google/xls/commit/e69f82f3299a9ab6e79f7f108fc3d8096fdb3c73 did a bunch of it but there was something remaining to do finish this up, Mark can you remind me what that was?

cdleary, 2021-03-01:
Also will this allow us to close https://github.com/google/xls/issues/235 ?

cdleary, 2021-03-15:
Also we should add a brief section on packed vs unpacked arrays and how to think about / handle them (e.g. for people doing array based op codegen in the future) to https://github.com/google/xls/blob/main/docs_src/vast.md -- maybe also can note they're all uplinked to VerilogFile while we're in there.

cdleary, 2021-03-15:
> I believe [e69f82f](https://github.com/google/xls/commit/e69f82f3299a9ab6e79f7f108fc3d8096fdb3c73) did a bunch of it but there was something remaining to do finish this up, Mark can you remind me what that was?

Status here is that we're not generating packed arrays yet, but all the infrastructure is present in VAST.

meheff, 2022-08-29:
Motivating example, consider an array of XLS type bits[16][3][2]. Currently this is declared as an unpacked array like so:

```
wire [15:0] foo[2][3];
```

We should emit this as:

```
wired [15:0][2:0][1:0] foo;
```

A problem is that plain Verilog does not support packed arrays so in Verilog arrays might be represented as a flat bit vector.