google/xls #332: Try replacing Ford-Fulkerson min-cut with spectral min-cut

taktoa, 2021-03-11:
There is a much simpler algorithm for computing min-cut of a graph than the Ford-Fulkerson implementation in `xls/data_structures/min_cut.cc`.

The [idea](https://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.592.1730&rep=rep1&type=pdf) is to compute the eigenvalues and eigenvectors of the graph Laplacian (`L(G) = D(G) - A(G)` where `D(G)` is the degree matrix, which is a diagonal matrix with the degree of each node along the diagonal, and `A(G)` is the adjacency matrix), and then find the eigenvector corresponding to the second smallest eigenvalue, called the Fiedler vector. Then each element of the Fiedler vector tells you which side of the cut a node will end up on -- if the Fiedler vector is `f`, then the `k`th node will be in part A if `f[k] < 0` and part B if `f[k] > 0`.

It is also possible the algorithm will be faster when implemented this way, due to data parallelism and regular memory accesses. And this technique is [more flexible](https://snap-stanford.github.io/cs224w-notes/network-methods/spectral-clustering) than the Ford-Fulkerson cut, since with some modification it can allow multi-way partitioning, for example.

meheff, 2021-03-25:
TIL: spectral graph partitioning. That is pretty wild! It'd be interesting to see if it can be applied to our multi-way partitioning problem of breaking a graph into stages. Wonder how the delay constraints would fit in (e.g, node N must be scheduled in stages 1 or 2). Currently, since we essentially do one stage cut at a time we can preassign these nodes with constraints to one partition or the other (e.g, for node N if we're performing the cut between stage 2 and stage 3 we known node N must be in the first partition because it must be scheduled in stage 1 or 2).

taktoa, 2021-04-30:
Realized that we actually need minimum s,t-cuts in the scheduler, and I don't think the spectral algorithm is amenable to those.

closed 2021-04-30 by taktoa