google/xls #338: Applying external functions with map builtin broken in interpreter

jbaileyhandle, 2021-03-15:
The following code used to work (float::unflatten is imported from another file):
```
#![test]
fn test_fft() {
  ...
  let real_in: F32[4] = map(u32[4]:[1,2,3,4], float32::unflatten);
  ...
 }
```
Now, this produces the following error:
```
[ RUN UNITTEST  ] test_fft
E0315 20:27:23.176871      16 interpreter.cc:319] INTERNAL: XLS_RET_CHECK failure (xls/dslx/interpreter.cc:319) fn_data.function->owner() == current_type_info_->module() (0x55ee591558a0 vs. 0x55ee5914d0c0) 
0x55ee55061d9a: absl::internal_statusor::StatusOrData<>::StatusOrData<>()
0x55ee550637f6: xls::dslx::Interpreter::CallFnValue()
0x55ee55063840: xls::dslx::AbstractInterpreterAdapter::CallValue()
0x55ee5507205b: xls::dslx::BuiltinMap()
0x55ee55061a83: xls::dslx::Interpreter::RunBuiltin()
0x55ee5506379e: xls::dslx::Interpreter::CallFnValue()
0x55ee550682e3: xls::dslx::Interpreter::EvaluateInvocation()
0x55ee55068c7e: xls::dslx::Evaluator::HandleInvocation()
0x55ee55061f43: xls::dslx::Interpreter::Evaluate()
0x55ee5506238d: xls::dslx::AbstractInterpreterAdapter::Eval()
0x55ee5508bac8: xls::dslx::EvaluateLet()
0x55ee5505fd9b: xls::dslx::Evaluator::HandleLet()
0x55ee55061f43: xls::dslx::Interpreter::Evaluate()
0x55ee5506759f: xls::dslx::Interpreter::RunTest()
0x55ee54fcfecb: xls::dslx::(anonymous namespace)::ParseAndTest()
0x55ee54f687d5: main
0x7f91e44230b3: __libc_start_main

E0315 20:27:23.176933      16 interpreter.cc:179] error @ xls/third_party/machsuite/fft/fft.x:136:28-136:66: INTERNAL: XLS_RET_CHECK failure (xls/dslx/interpreter.cc:319) fn_data.function->owner() == current_type_info_->module() (0x55ee591558a0 vs. 0x55ee5914d0c0) 
E0315 20:27:23.176946      16 interpreter.cc:179] error @ xls/third_party/machsuite/fft/fft.x:136:3-159:1: INTERNAL: XLS_RET_CHECK failure (xls/dslx/interpreter.cc:319) fn_data.function->owner() == current_type_info_->module() (0x55ee591558a0 vs. 0x55ee5914d0c0) 
E0315 20:27:23.176983      16 interpreter_main.cc:153] Internal error: INTERNAL: XLS_RET_CHECK failure (xls/dslx/interpreter.cc:319) fn_data.function->owner() == current_type_info_->module() (0x55ee591558a0 vs. 0x55ee5914d0c0) 
[        FAILED ] test_fft: internal error: INTERNAL: XLS_RET_CHECK failure (xls/dslx/interpreter.cc:319) fn_data.function->owner() == current_type_info_->module() (0x55ee591558a0 vs. 0x55ee5914d0c0) 
```

The issue goes away if we add a local function that calls float32::unflatten:
```
fn my_unflatten(myin: u32) -> F32 {
  float32::unflatten(myin)
}

#![test]
fn test_fft() {
  ...
  let real_in: F32[4] = map(u32[4]:[1,2,3,4], my_unflatten);
  ...
 }
```

jbaileyhandle, 2021-03-15:
Strangely, the following runs with no problem:
```
let samples = map(s32[6]:[1, 2, 3, 4, 5, 6], float32::cast_from_fixed);
```
where float32::cast_from_fixed is a yet-to-be-committed-to-main function defined in the same place as float32:unflatten (apfloat) and imported in the same way.  The above code and a similar line of code using float32::unflatten were tested in the same file, and the issue only showed up for unflatten.  So, this isn't obviously some sort of build or import issue.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): The removed legacy AST interpreter (`Interpreter`) was replaced by the bytecode interpreter (`xls/dslx/bytecode/bytecode_interpreter.cc`), which supports `map` with imported functions as tested in `xls/dslx/tests/map_multi_module.x`, `xls/dslx/tests/map_of_stdlib_parametric.x`, and `xls/dslx/type_system/typecheck_module_test.cc:838`. (Closed by Gemini)

closed 2026-11-14 by allight