google/xls #344: Compile DSL to executable

cg31, 2021-03-19:
It seems the DSL is only interpreted now.

Is there a plan to compile it as optimized executable and run alone?

One purpose for doing it is to write a model in DSL directly, not in C/C++.

cdleary, 2021-03-20:
Hi @cg31 thanks for reaching out!

Right now the DSL is just JIT (Just in Time) compiled, and we've mostly expected it to be integrated into a broader modeling environment (like we currently do for a few different simulators). We have a build macro that creates a header so you can invoke DSL functions as if they were C++ functions (JIT compiling the entry point on first invocation), see: https://google.github.io/xls/build_system/#dslx_jit_wrapper

This makes C++ APIs that intend to have the kind of signature you'd expect; e.g. see this usage for the FP32 adder:

https://github.com/google/xls/blob/main/xls/modules/fpadd_2x32_test.cc#L64

That being said, I think this issue flags that we could consider an AoT (ahead of time) compilation process that also linked in enough runtime. If we did that, I imagine we'd need to decide what I/O stimulus usefully looked like with respect to the outside world. Thoughts on that, or what you might be looking for in that sense? Something on the easier side would be "turn this unit test entry point into a binary", but presumably we want binaries we can provide stimulus to on the fly.

cdleary, 2021-03-20:
Oh, and I realize the original question may implicitly be, "can I expect my DSL code to run at native speeds [like C++ HLS would] when I run it JIT compiled", and from what we've seen so far the answer seems to be yes, but more data points are always welcome.

cg31, 2021-03-20:
Currently, when a comlex processor (for example, GPU or processor for ML, like google's TPU) is designed, it is common to write a model in C/C++ first, to verify the design for correctness and maybe performance, then write RTL by hands.

My thought is if we can write this model in DSL, then we don't need to to write RTL, which is generated from Model directly.

In the same time, if DSL can be compiled to native code like amd64, we can build the model  into Qemu, to verify it as a device.

cdleary, 2021-03-20:
Right, components written in the DSL do currently integrate _into_ that simulator! We are currently expecting the outer scaffolding (e.g. what defines the virtual tick for the overall simulation, and has some mechanism for interacting with the outside world outside of the simulation) to be written in some other language, such as C++. The DSL is fully synthesizable at the moment, and so is notably not a replacement for a general purpose host language where things like malloc, syscalls, and virtual dispatch are possible. (This is part of what makes it domain specific.)

We're trying to make bigger and bigger components expressable in the DSL over time, and we at least have demos of dropping a "device function" into a standard "uncore shell" so that you can get push-button testing of some gateware-accelerated function, but it's still a far cry from any "push button, receive a full chip all written in the DSL" kind of flow, it just knows enough about how to get some I/O stimulus from that uncore shell to try out your gateware.

To your last point, you could, today, make a shared object target that ran DSL functionality and link it into qemu. I think that's an excellent idea for a demo flow, kind of similar to the device function prototyping flow I mentioned above but all with software simulation.

Does it make sense for me to think of this issue as "Demo flow of integrating DSL-written component into QEMU?", does that capture the main idea you were thinking?

cg31, 2021-03-20:
@cdleary, all your explanations make sense, and thanks for your reponses.

It looks like an extra layer in C/C++ with DSL model running in JIT could achieve what I want.

RobSpringer, 2022-05-06:
I've got some WIP to enable AOT compilation (I sure do love TLAs!), and I'm going to claim this issue for it.

commit 90570831 (2022-05-11): Adding sort-of unit test for the AOT compiler.

Also fixed a bug it found: the global objects could collide if we're not namespacing the exported function, so we need to wrap them in a private namespace.

Resolves #344.

PiperOrigin-RevId: 448090657
closed 2022-05-11 by copybara-service[bot]