google/xls #359: Type alias name duplication

vincent-mirian-google, 2021-03-26:
For input
```
type xType = u8;
type xType = u64;
  
fn main() -> xType {
  xType: 4
}

#![test]
fn main_test() {
  let x: xType = main();
  let _ = assert_eq(x, u8:4);
  ()
}
```

received error:
```
  0025: fn main_test() {
  0026:   let x: xType = main();
* 0027:   let _ = assert_eq(x, u8:4);
  ~~~~~~~~~~~~~~~~~~~~~~~~~^-------^ XlsTypeError: uN[64] vs uN[8]: Want arguments 0 and 1 to 'assert_eq' to be of the same type; got uN[64] vs uN[8]
  0028:   ()
  0029: }
```

A warning message or documentation is required when for name reference is duplicated for type alias. I would prefer an error when name references are duplicated. What are your thoughts @cdleary ? Thanks,

cdleary, 2021-03-26:
Yes, we should be flagging this name getting re-bound, type aliases getting bound just once at the module scope makes sense to me -- thanks for flagging / filing!

RobSpringer, 2022-04-26:
Fixed by the above patch.

closed 2022-04-26 by RobSpringer