google/xls #373: JIT compiled code crashes, crasher 2b5f

meheff, 2021-04-05:
Crasher will be checked in shortly. Curious the minimized IR sample just fails with JIT miscompare:

```
package sample

fn __sample__main(x0: bits[50][9], x2: bits[5], x3: bits[18], x4: bits[51], x5: bits[30], x6: bits[58]) -> bits[51] {
  sign_ext.8: bits[62] = sign_ext(x3, new_bit_count=62, id=8)
  x7: bits[62] = literal(value=2305843009213693951, id=7, pos=0,6,20)
  x8: bits[62] = add(sign_ext.8, x7, id=9, pos=0,7,32)
  x9: bits[1] = dynamic_bit_slice(x7, x8, width=1, id=10, pos=0,8,21)
  x23: bits[1] = not(x9, id=31, pos=0,20,18)
  ret x36: bits[51] = zero_ext(x23, new_bit_count=51, id=57, pos=0,32,22)
}
```

```
$ xls/tools/eval_ir_main --input="[bits[50]:0x2_0c7a_c27b_22c4, bits[50]:0x1_ffff_ffff_ffff, bits[50]:0x2_f52c_0057_9850, bits[50]:0x1_ffff_ffff_ffff, bits[50]:0x3_ffff_ffff_ffff, bits[50]:0x1_5555_5555_5555, bits[50]:0x0, bits[50]:0x1_5555_5555_5555, bits[50]:0x3_ffff_ffff_ffff]; bits[5]:0x2; bits[18]:0x1; bits[51]:0x5_5555_5555_5555; bits[30]:0x3fff_ffff; bits[58]:0x132_9fff_d3ef_7cfb" minimized.ir --test_llvm_jit
bits[51]:0x1
bits[51]:0x7_ffff_ffff_ffff
F0405 14:35:09.209457  647321 eval_ir_main.cc:368] Check failed: ::absl::OkStatus() == (xls::RealMain(positional_arguments[0])) (OK vs. INVALID_ARGUMENT: Miscompare for input "[bits[50]:0x2_0c7a_c27b_22c4, bits[50]:0x1_ffff_ffff_ffff, bits[50]:0x2_f52c_0057_9850, bits[50]:0x1_ffff_ffff_ffff, bits[50]:0x3_ffff_ffff_ffff, bits[50]:0x1_5555_5555_5555, bits[50]:0x0, bits[50]:0x1_5555_5555_5555, bits[50]:0x3_ffff_ffff_ffff]; bits[5]:0x2; bits[18]:0x1; bits[51]:0x5_5555_5555_5555; bits[30]:0x3fff_ffff; bits[58]:0x132_9fff_d3ef_7cfb"
  JIT: bits[51]:0x7_ffff_ffff_ffff
  interpreter: bits[51]:0x1) 
```

meheff, 2021-04-05:
Minimized sample below. Repro:

```
/xls/tools/eval_ir_main --random_inputs=1 foo.ir
*** SIGBUS (@(nil)) ***
PC: @     0x7fc97116705a  (unknown)  (unknown)
    @     0x5569bed5bab6        960  FailureSignalHandler()
    @     0x7fc9719959a0  830977856  (unknown)
    @     0x5569bd507ad1        416  xls::(anonymous namespace)::Eval()
    @     0x5569bd503774        848  xls::(anonymous namespace)::RealMain()
    @     0x5569bd501677        368  main
    @     0x7fc971803bbd        208  __libc_start_main
    @     0x5569bd501029  (unknown)  ../sysdeps/x86_64/start.S:108 _start
```

```
package sample

fn __sample__main(x3: bits[18]) -> bits[51] {
  sign_ext.8: bits[62] = sign_ext(x3, new_bit_count=62, id=8)
  x7: bits[62] = literal(value=2305843009213693951, id=7, pos=0,6,20)
  x7__2: bits[62] = literal(value=2305843009213693951, id=66, pos=0,6,20)
  x8: bits[62] = add(sign_ext.8, x7, id=9, pos=0,7,32)
  x9: bits[1] = dynamic_bit_slice(x7__2, x8, width=1, id=10, pos=0,8,21)
  x17: bits[51][4] = literal(value=[0, 0, 0, 0], id=57, pos=0,14,20)
  x23: bits[1] = not(x9, id=31, pos=0,20,18)
  ret x36: bits[51] = array_index(x17, indices=[x23], id=50, pos=0,32,22)
}
```

meheff, 2021-04-05:
Given the following llvm code (this is the unoptimized code we generate):

```
; ModuleID = 'the_module'
source_filename = "the_module"
target datalayout = "e-m:e-p270:32:32-p271:32:32-p272:64:64-i64:64-f80:128-n8:16:32:64-S128"

define void @"sample::__sample__main"([2 x i8*]* %0, i2* %1, i64 %2, i64 %3) {
so_basic:
  %4 = getelementptr [2 x i8*], [2 x i8*]* %0, i64 0, i64 0
  %5 = load i8*, i8** %4, align 8
  %6 = bitcast i8* %5 to i46*
  %x0 = load i46, i46* %6, align 8
  %7 = getelementptr [2 x i8*], [2 x i8*]* %0, i64 0, i64 1
  %8 = load i8*, i8** %7, align 8
  %9 = bitcast i8* %8 to i42*
  %x1 = load i42, i42* %9, align 8
  %x7 = call i8 @"sample::____sample__main_counted_for_0_body"(i4 0, i8 -1, i64 %3)
  %10 = icmp uge i8 %x7, 8
  %11 = select i1 %10, i8 0, i8 %x7
  %12 = lshr i8 -1, %11
  %13 = trunc i8 %12 to i1
  %x13 = select i1 %10, i1 false, i1 %13
  %14 = icmp eq i1 %x13, true
  %15 = select i1 %14, i2 -1, i2 0
  %x17 = or i2 0, %15
  %16 = lshr i1 %x13, true
  %x27 = add i2 %x17, 1
  store i2 %x27, i2* %1, align 1
  ret void
}

define i8 @"sample::____sample__main_counted_for_0_body"(i4 %i, i8 %x, i64 %0) {
so_basic:
  ret i8 %x

so_basic1:                                        ; No predecessors!
  ret i8 %x
}
```

The JIT optimizes it to:

```
; ModuleID = 'the_module'
source_filename = "the_module"
target datalayout = "e-m:e-p270:32:32-p271:32:32-p272:64:64-i64:64-f80:128-n8:16:32:64-S128"

; Function Attrs: nofree norecurse nounwind willreturn writeonly
define void @"sample::__sample__main"([2 x i8*]* nocapture readnone %0, i2* nocapture %1, i64 %2, i64 %3) local_unnamed_addr #0 {
so_basic:
  store i2 -1, i2* %1, align 1
  ret void
}
```

While command line LLVM opt tool produces:

```; ModuleID = '/tmp/bad2.ll'
source_filename = "the_module"
target datalayout = "e-m:e-p270:32:32-p271:32:32-p272:64:64-i64:64-f80:128-n8:16:32:64-S128"

; Function Attrs: nofree norecurse nounwind willreturn writeonly
define void @"sample::__sample__main"([2 x i8*]* nocapture readnone %0, i2* nocapture %1, i64 %2, i64 %3) local_unnamed_addr #0 {
so_basic:
  store i2 1, i2* %1, align 1
  ret void
}

; Function Attrs: norecurse nosync nounwind readnone willreturn
define i8 @"sample::____sample__main_counted_for_0_body"(i4 %i, i8 returned %x, i64 %0) local_unnamed_addr #1 {
so_basic:
  ret i8 %x
}
```

Note the different value being stored 1 (bits[2]:1) vs -1 (bits[2]:3). This is the same value mismatch eval_ir_reports (returning 1 is correct).

allight, 2026-11-10:
This crasher (crasher_2021-04-05_2b5f.x) is no longer in the known-failing lists of //xls/fuzzer:regression_tests, so it appears to be fixed (re-enabled in https://github.com/google/xls/commit/ecc85366c6e728e54a8811edfa6c478a71a2759d).

closed 2026-11-10 by allight