google/xls #384: Cannot use XLS binaries out-of-tree

ted-xie, 2021-04-12:
repro:

```
cp $XLS_HOME/bazel-bin/xls/dslx/ir_converter_main /path/to/somewhere/else
/path/to/somewhere/else/ir_converter_main
```

Result:
```
E0411 17:33:22.710784   13597 get_runfile_path.cc:46] INTERNAL: XLS_RET_CHECK failure (xls/common/file/get_runfile_path.cc:46) runfiles != nullptr Failed to initialize Runfiles: ERROR: external/bazel_tools/tools/cpp/runfiles/runfiles.cc(119): cannot find runfiles (argv0="bin/ir_converter_main")
0x5560631f73ef: xabsl::StatusBuilder::CreateStatusAndConditionallyLog()
0x556062d4bdf1: xabsl::StatusBuilder::operator absl::lts_20210324::Status()
0x5560631d6441: absl::lts_20210324::internal_statusor::StatusOrData<>::StatusOrData<>()
0x5560631d6162: absl::lts_20210324::StatusOr<>::StatusOr<>()
0x5560631d5b3d: xls::(anonymous namespace)::GetRunfiles()
0x5560631d5f15: xls::InitRunfilesDir()
0x5560631d4907: xls::InitXls()
0x556062d4add9: main
0x7fa02544c0b3: __libc_start_main

F0411 17:33:22.710890   13597 init_xls.cc:30] Check failed: ::absl::OkStatus() == (InitRunfilesDir(argv[0])) (OK vs. INTERNAL: XLS_RET_CHECK failure (xls/common/file/get_runfile_path.cc:46) runfiles != nullptr Failed to initialize Runfiles: ERROR: external/bazel_tools/tools/cpp/runfiles/runfiles.cc(119): cannot find runfiles (argv0="bin/ir_converter_main")) 
*** Check failure stack trace: ***
0x5560632019cd: xls::logging_internal::LogMessage::SendToLog()
0x556063200cff: xls::logging_internal::LogMessage::Flush()
0x556063201b2a: xls::logging_internal::LogMessageFatal::~LogMessageFatal()
0x5560631d49e4: xls::InitXls()
0x556062d4add9: main
0x7fa02544c0b3: __libc_start_main
Aborted (core dumped)

```

XLS binaries should support out-of-tree execution.

git hash is `14d5f6e7ba1fd553cdfc9a0d5cb772c1c3ea75cf`

cdleary, 2022-10-28:
CC @proppy as this relates to packaging work too

hongted, 2022-10-28:
The runfiles directory also needs to be copied.  

```
cd $XLS_HOME
./xls/tools/make_release.py /path/to/somewhere/else
ln -s /path/to/somewhere/else/xls/dslx/ir_converter_main  /path/to/somewhere/else/ir_converter_main
```

Will create  copy all dependencies and create a symlink.

proppy, 2026-10-28:
I believe this is now fixed with the releases: https://github.com/google/xls/releases

closed 2026-10-28 by proppy