google/xls #385: XLS codegen emits invalid SystemVerilog code

ted-xie, 2021-04-12:
See my  DSLX design [here](https://github.com/ted-xie/xls_regex/commit/a988738cc7eb1c5869159a4cda301f83f920e5cb).

```
cd src_dslx
XLS_HOME=/path/to/xls make donut_opt.v # takes about 20 seconds
yosys -p 'read_verilog -sv donut_opt.v; synth_xilinx'
```
It looks like the XLS codegen tries to generate array-ish datatypes in the C++ style rather than Verilog style, for example `wire literal_59710[256];` vs `wire [255:0] literal_59710;`.

Full diff [can be seen here](https://gist.github.com/ted-xie/db6f5d9f96cb7e9ca49e724ae23d13d1).

githash: `14d5f6e7ba1fd553cdfc9a0d5cb772c1c3ea75cf`

ted-xie, 2021-04-15:
For now, disabling SystemVerilog generation sidesteps the syntax error. I'm going to keep this bug open though, because SystemVerilog syntax validity is important.

meheff, 2021-04-20:
What simulator gave you an error?

I believe the following is valid SystemVerilog:

wire foo[123];

It's an unpacked array of scalars. And regarding the apostrophe:

assign foo = '{0, 1, ... };

That's an array assign pattern described in 10.9.1 of the SystemVerilog LRM 1800-2017.

FWIW, we plan on moving away from unpacked arrays for the array representation in SystemVerilog. We'll use packed arrays in like so:

wire [7:0][122:0] foo; // array of 123 elements of width 8.

And we'll use flattened vectors in Verilog where packed arrays are not supported:

wire [983:0] foo; // Width of 984 is 8 * 123.

meheff, 2021-07-20:
Ted, given my previous comment, is this still an issue?

ted-xie, 2022-04-15:
Sorry for the late reply, I never saw the notification for your comment.

I understand it's part of the LRM, but it's also not the standard way that 1-dimensional bit vectors are expressed in SystemVerilog (in fact, many linters would count `wire x[255:0]` as a lint error). Usually it's expressed the opposite way: `wire [255:0] x`. I think the assignment syntax for the two coding styles is identical.

meheff, 2022-05-12:
Switching to using packed arrays will resolve this issue and is something we want to do: https://github.com/google/xls/issues/320

Closing this a s duplicate

meheff, 2022-05-12:
Duplicate of #320

closed 2022-05-12 by meheff