google/xls #392: Serialize DSLX Type Information

vincent-mirian-google, 2021-04-14:
In the bazel rule, dslx_library, serialize type information from parse and type check task for use by other rules to avoid recomputing type information.

For example:
```
        # Serialize type information from parse and type check task so it can be used by target c.
        dslx_library(
            name = "a",
            srcs = [
                "a.x",
            ],
        )

        # Depends on target a.
        # Use serialized type information from target a to avoid recomputing the parsing and type checking task of a.
        dslx_library(
            name = "c",
            srcs = [
                "c.x",
            ],
            deps = [
                ":a",
            ],
        )
```
Reference: https://github.com/google/xls/issues/311

vincent-mirian-google, 2021-04-14:
Some notes for design:
- dump TypeInfo/DeduceCtx instances containing bindings of reference to type 
- design a protobuf representation that preserves TypeInfo/DeduceCtx information

vincent-mirian-google, 2021-11-20:
Additional note: When https://github.com/google/xls/issues/521 is implemented, the DSLX interpreter would need to scan for duplicate configurations of the same DSLX source file.

cdleary, 2022-01-27:
Note that #521 is now resolved.