google/xls #394: [Feature-Request] Improve Verilog codegen for tuple outputs

ted-xie, 2021-04-15:
I have a DSLX function that returns a tuple of type `(u1, u8)`. The tuple fields are called `(result, state)`. I would expect the emitted verilog to look like this:

```
output wire result,
output wire [7:0] state
```

Instead, I get a packed bit vector called `output wire [8:0] out`.

This is still usable, but generally it would be easier to integrate the generated code if the fields were unpacked into separate output ports.

vincent-mirian-google, 2021-04-15:
A brief comment: perhaps, multiple return values may be a more general solution?

cdleary, 2021-04-28:
cc @meheff I think the reworking he's doing of codegen pipeline could tie in with how tuples or records (named tuples) end up presented on the module signature. Mark thoughts on where this one will fit? (I thought we had another issue open for this but can't find it!) This relates to google/xls#195

allight, 2024-06-11:
We might want to build off of the work done on https://github.com/google/xls/issues/1239 to support this.

We could have an annotation on structs/etc `#[verilog_export("name")]`.

We would need to decide how to pass this down to the codegen. The easiest way might be to have the ir_convert do the verilog generation of the struct and pass it down in a field in the ir-interface proto. This is sort of ugly however.

@proppy any ideas?

proppy, 2024-06-11:
@allight I was thinking of something like what was suggested in #1433 (as a follow-up to #1429) where we have `fn` annotation to name the return argument, ex:
```
#[output_port_name(foo, bar, hop)]
fn user_module() -> (u1, u2, u3) {
}
```
or
```
```
#[named_return]
fn user_module() -> (foo: u1, bar: u2, hop: u3) {
}
```
would codegen:
```
module user_module(
    output wire foo,
    output wire [1:0] bar,
    output wire [2:0] hop
);
```

allight, 2026-11-14:
Keeping this issue open (reviewed by @allight): While `CodegenOptions::split_outputs` (`xls/codegen/codegen_options.h:249-250`, `xls/codegen/block_conversion.cc:363,630`) splits tuple outputs into indexed ports (`out_0`, `out_1`, ...), meaningful port names and structure are important, so we are not closing this issue. (Summarized by Gemini)