google/xls #398: interpreter_main get 'absl::BadStatusOrAccess' error

pavel532, 2021-04-19:
Hello!

I have the following .x code:
```rust
const BLOCK_SIZE       = s64:64;
const BLOCK_LINE_COUNT = s64:8;
const BLOCK_COL_COUNT  = s64:8;

import float32
import xls.modules.fpadd_2x32

type F32 = float32::F32;

fn compress(pixels: s16[BLOCK_SIZE],
            compression_quality: s16) -> u16[BLOCK_SIZE]
{

//  let BLOCK_LINE_COUNT_my = u16:8;

    let temp_matrix = for(i, temp_matrix) : (u16, F32[BLOCK_SIZE]) in range (u16:0, BLOCK_LINE_COUNT as u16)
    {
        for(j, temp_matrix_j) : (u16, F32[BLOCK_SIZE]) in range (u16:0, BLOCK_COL_COUNT as u16)
          {
              let temp_value = for(k, temp_value_k) : (u16, F32) in range (u16:0, BLOCK_COL_COUNT as u16)
              {
                  let pixels_ik_f64 = F32::one(u1:0);  //  <- this line
                  float32::one(u1:1)
              } (float32::zero(u1:0));
                F32[BLOCK_SIZE]:[float32::zero(u1:0), ...]
          } (temp_matrix)

      } (F32[BLOCK_SIZE] : [float32::zero(u1:0) , ...]);

    u16[BLOCK_SIZE]: [0, ...]
}
```

interpreter_main failes on this .x file:

$ <PATH_TO_INTERPRETER>/interpreter_main <PATH_TO_X>/good_file.x
```
E0419 19:28:53.489329   11441 deduce.cc:776] INTERNAL: XLS_RET_CHECK failure (xls/dslx/deduce.cc:776) enum_type != nullptr 
0x55f1f32e4752: absl::internal_statusor::StatusOrData<>::StatusOrData<>()
0x55f1f33051ee: xls::dslx::(anonymous namespace)::DeduceColonRef()
0x55f1f3305b66: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleColonRef()
0x55f1f33cf421: xls::dslx::ColonRef::Accept()
0x55f1f32ffbce: xls::dslx::Deduce()
0x55f1f32d311c: std::_Function_handler<>::_M_invoke()
0x55f1f32d4057: xls::dslx::DeduceCtx::Deduce()
0x55f1f3305f0b: xls::dslx::(anonymous namespace)::DeduceInvocation()
0x55f1f3308456: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleInvocation()
0x55f1f33cf621: xls::dslx::Invocation::Accept()
0x55f1f32ffbce: xls::dslx::Deduce()
0x55f1f32d311c: std::_Function_handler<>::_M_invoke()
0x55f1f32e5816: xls::dslx::DeduceAndResolve()
0x55f1f32f4451: xls::dslx::(anonymous namespace)::DeduceLet()
0x55f1f32f4a06: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleLet()
0x55f1f33cf9e1: xls::dslx::Let::Accept()
0x55f1f32ffbce: xls::dslx::Deduce()
0x55f1f32d311c: std::_Function_handler<>::_M_invoke()
0x55f1f32e5816: xls::dslx::DeduceAndResolve()
0x55f1f32f4c8c: xls::dslx::(anonymous namespace)::DeduceFor()
0x55f1f32f51b6: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleFor()
0x55f1f33cf911: xls::dslx::For::Accept()
0x55f1f32ffbce: xls::dslx::Deduce()
0x55f1f32d311c: std::_Function_handler<>::_M_invoke()
0x55f1f32e5816: xls::dslx::DeduceAndResolve()
0x55f1f32f4451: xls::dslx::(anonymous namespace)::DeduceLet()
0x55f1f32f4a06: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleLet()
0x55f1f33cf9e1: xls::dslx::Let::Accept()
0x55f1f32ffbce: xls::dslx::Deduce()
0x55f1f32d311c: std::_Function_handler<>::_M_invoke()
0x55f1f32e5816: xls::dslx::DeduceAndResolve()
0x55f1f32f4c8c: xls::dslx::(anonymous namespace)::DeduceFor()
0x55f1f32f51b6: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleFor()
0x55f1f33cf911: xls::dslx::For::Accept()
0x55f1f32ffbce: xls::dslx::Deduce()
0x55f1f32d311c: std::_Function_handler<>::_M_invoke()
0x55f1f32e5816: xls::dslx::DeduceAndResolve()
0x55f1f32f4c8c: xls::dslx::(anonymous namespace)::DeduceFor()
0x55f1f32f51b6: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleFor()
0x55f1f33cf911: xls::dslx::For::Accept()
0x55f1f32ffbce: xls::dslx::Deduce()
0x55f1f32d311c: std::_Function_handler<>::_M_invoke()
0x55f1f32e5816: xls::dslx::DeduceAndResolve()
0x55f1f32f4451: xls::dslx::(anonymous namespace)::DeduceLet()
0x55f1f32f4a06: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleLet()
0x55f1f33cf9e1: xls::dslx::Let::Accept()
0x55f1f32ffbce: xls::dslx::Deduce()
0x55f1f32d311c: std::_Function_handler<>::_M_invoke()
0x55f1f32e5816: xls::dslx::DeduceAndResolve()
0x55f1f32d825e: xls::dslx::CheckFunction()

terminate called after throwing an instance of 'absl::BadStatusOrAccess'
  what():  Bad StatusOr access
Aborted (core dumped)
```

but if comment out line 
```
 let pixels_ik_f64 = F32::one(u1:0);
```
then interpreter_main run without any error.

cdleary, 2021-04-19:
Thanks for the report! Definitely an internal error, will check that out and make a regression test, will update this issue with updates.

closed 2021-04-19 by copybara-service[bot]

cdleary, 2021-04-19:
Should have been giving an error message, should be giving it appropriately now, thanks again for the report!