google/xls #399: cast u8 to float32

pavel532, 2021-04-19:
Hello!
Could you realize cast from u8 to float32.

example:
```rust
import float32
import xls.modules.fpadd_2x32

type F32 = float32::F32;

fn cast_u8_F32 (x:u8) -> u8
{
    let a = u8:1;
    let b = a as F32;
    let c = fpadd_2x32::fpadd_2x32(b, float32::one(u1:0));

    u8:1
}
```

$ <PATH_TO_INTERPRETER>/interpreter_main <PATH_TO_X>/test_cast_f32.x

```sh
F0419 20:12:24.393671   12480 interpreter_main.cc:297] Check failed: ::absl::OkStatus() == (status) (OK vs. INVALID_ARGUMENT: Could not build IR: Expected operand 0 of invoke.307 to have type (bits[1], bits[8], bits[23]), has type bits[32]. File: 0, Line: 9, Col: 34
Stack Trace:
0x555d0ca0b3c5: xls::BuilderBase::Invoke()
0x555d0c88709d: std::_Function_handler<>::_M_invoke()
0x555d0c88994d: std::_Function_handler<>::_M_invoke()
0x555d0c899ca6: xls::dslx::(anonymous namespace)::FunctionConverter::DefWithStatus()
0x555d0c89a052: xls::dslx::(anonymous namespace)::FunctionConverter::Def()
0x555d0c8a8b5f: xls::dslx::(anonymous namespace)::FunctionConverter::HandleInvocation()
0x555d0c8aa6a6: xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleInvocation()
0x555d0c9ba621: xls::dslx::Invocation::Accept()
0x555d0c8b1e0c: xls::dslx::(anonymous namespace)::FunctionConverter::HandleLet()
0x555d0c8b2616: xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
0x555d0c9ba9e1: xls::dslx::Let::Accept()
0x555d0c8998dc: xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
0x555d0c8b2305: xls::dslx::(anonymous namespace)::FunctionConverter::HandleLet()
0x555d0c8b2616: xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
0x555d0c9ba9e1: xls::dslx::Let::Accept()
0x555d0c8998dc: xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
0x555d0c8b2305: xls::dslx::(anonymous namespace)::FunctionConverter::HandleLet()
0x555d0c8b2616: xls::dslx::(anonymous namespace)::FunctionConverterVisitor::HandleLet()
0x555d0c9ba9e1: xls::dslx::Let::Accept()
0x555d0c8998dc: xls::dslx::(anonymous namespace)::FunctionConverterVisitor::Visit()
0x555d0c8ab736: xls::dslx::(anonymous namespace)::FunctionConverter::HandleFunction()
0x555d0c8ac5b7: xls::dslx::(anonymous namespace)::ConvertOneFunctionInternal()
0x555d0c8ad083: xls::dslx::ConvertModuleToPackage()
0x555d0c8819ff: xls::dslx::(anonymous namespace)::ParseAndTest()
0x555d0c81b541: main
0x7fc302d94bf7: __libc_start_main
) 
```

Сan you suggest a workaround ?

cdleary, 2021-04-20:
Hi! I think maybe what you're looking for is the `cast_from_fixed` operator here:

https://github.com/google/xls/blob/864f6cabdbdebb5fa9d1ffe3832212c9927fd987/xls/dslx/stdlib/apfloat.x#L186

But we should also make a better error message for attempted conversion of an integral value to a struct than what you're seeing.

closed 2021-04-20 by copybara-service[bot]

cdleary, 2021-04-20:
Closing this on account of the error message being fixed, but let us know if the cast operation I cited doesn't work for you!

pavel532, 2021-04-20:
Thanks a lot for a quick answer
I will check this later.