google/xls #401: Question: for loop in DSLX

pavel532, 2021-04-21:
Hello!

Please, say, how to write the following C code in DSLX

```cpp
        while (nibble_count > 1) {
            output[byte_count++] = (nibbles[0] << 4) + nibbles[1];
            nibbles[0] = nibbles[2];
            nibbles[1] = nibbles[3];
            nibbles[2] = nibbles[4];
            nibbles[3] = nibbles[5];
            nibble_count -= 2;
        }
```
The problem is that I saw in the DSLX language only loops based on range () - there the iteration goes in ascending order, with increasing the counter by 1

cdleary, 2021-04-22:
Hi @pavel532 -- to make this a hardware pipeline, the pipeline has to have an upper bound on the number of iterations (because pipelines are unrolled in space), so you would do something like:

```dslx
fn collapse_nibbles<NIBBLES: u32, BITS: u32 = NIBBLES*u32:4>(x: bits[BITS]) -> u8[NIBBLES] {
  let output_bytes_init = u8[NIBBLES]:[0, ...];
  for (i, output) in range(u32:0, NIBBLES) {
    let output_byte = (nibbles[u32:2*i] << 4) + nibbles[u32:2*i+u32:1];
    update(output, i, output_byte)
  }(output_bytes_init)
}
```

There is a "sequential emitter" generation mode that will make time multiplexed hardware, which could support an unbounded number of iterations using an FSM, but it's not currently exposed with a handy tool, but I'll try to see if we can expose that easily and update this issue.

There are some temporary limitations around expressing strides and converting the ranges to IR that I think we can prioritize given you're running into this. Thanks for the report! LMK if what I wrote above works for you, and please test it, I just wrote this into the text box without testing so may have bugs. :-)

cdleary, 2021-04-22:
(Note: marking as enhancement for the stride-by-two and its IR conversion, specifically.)

pavel532, 2021-04-30:
Thanks a lot for your answer!
But I implemented it a little differently.

cdleary, 2021-05-03:
@pavel532 Excellent! I think I realized afterwards I had a bug in what I typed but I forgot what it was, glad you were able to get your job done, that's the key. :-)

Leaving this open for striding enhancement.