google/xls #403: DSL should have trace_fmt!() builtin

cdleary, 2021-04-21:
User request -- right now the DSL only has a `trace!()`, and we only have a coarse grain mechanism for changing the format at the entire interpreter level as a kluge (see https://github.com/google/xls/commit/d90880e241649f1ad0d438d9e1eff76fcc25b5dd ) until we introduce a `trace_fmt!` macro.

Just starting with "{:d}" and "{:x}" and "{:b}" kind of thing would be fine for a first cut. We may want to limit support to stuff we can also eventually transpile into `$display` statements, but support for that is not the subject of this bug.

cdleary, 2021-04-26:
Note that the stopgap for this was https://github.com/google/xls/commit/d90880e241649f1ad0d438d9e1eff76fcc25b5dd -- Ravi said he'd be willing to take this on, thanks Ravi!

cg31, 2021-05-06:
It will be very useful, if we could use a formated trace to anayze the performace of a design.

I am trying to create a model in DSLX, it will have counters (of a global clock), like perf counters in OpenGL.

Formatted log is useful in DSLX interpreter and during JIT execution for development, but it is not needed in generated RTL, counters will be exposed and can be read in final product.

cdleary, 2022-05-10:
Ravi added this, e.g. in https://github.com/google/xls/commit/979a3d748b51bf741daca9590f330e9192ba5700#diff-ce11e7a7703efd6a70276859eccc9cd80f448b607b64ad2d128a1b63d05950f8 -- note that it is discarded in synthesis. I think it still needs docs though (there is a downstream design doc but I think we need some markdown) so leaving open for that, I'll try to write some up.

cdleary, 2022-05-10:
Another limitation I remember when writing up docs is that the JIT does not currently support `trace_fmt!`s that require data operands, so we should leave open for completing that as well.