google/xls #404: IR file created by FunctionBuilder won't parse in opt_main

spurserh, 2021-04-22:
blaze-bin/third_party/xls/tools/opt_main /tmp/test.ir
F0421 17:11:08.143368 1575568 opt_main.cc:88] Check failed: ::absl::OkStatus() == (xls::RealMain(positional_arguments[0])) (OK vs. INVALID_ARGUMENT: Expected token of type "ident" @ 3:37, but found: Token("keyword", value="next") @ 3:37; @ /tmp/test.ir)

spurserh, 2021-04-22:
Seems changing "next" to "nextc" allows this opt_main to succeed.

cdleary, 2021-04-22:
Good find, we should probably have a test that round trips this trying to use an identifier for each of the IR keywords:

https://github.com/google/xls/blob/f47a3a07863c47693bd4279969e5ca60173d2eb2/xls/ir/ir_scanner.h#L65

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `FunctionBase::GetIrReservedWords()` (`xls/ir/function_base.cc:642`) passes all IR scanner keywords from `Token::GetKeywords()` (`xls/ir/ir_scanner.h:72`, including `"next"`) to `node_name_uniquer_`, so `NameUniquer::SanitizeName` (`xls/ir/name_uniquer.cc:50`) prefixes reserved parameter and node names with `_` (`_next`) when emitting IR. (Closed by Gemini)

closed 2026-11-14 by allight