google/xls #405: A for loop with non-const upper bound

pavel532, 2021-04-22:
Hello!
If in the DSL file in the for loop, if I specify not a constant but a variable in the upper bound of the range, then the interpreter get the message:

```rust
#![test]
fn test_for()
{
    let a = u8:3;
//  const a = u8:3;
    let _ = for (i, acc) : (u8, u8) in range (u8:0, a)
    {
        acc + i
    }(u8:0);
    ()
}
```

```bash
$ <PATH_TO_interpreter_main>/interpreter_main <PATH_TO_X>/test_for.x

E0422 08:51:22.003500    6664 command_line_utils.cc:35] Could not extract a textual position from error message: NOT_FOUND: Could not find bindings entry for identifier: "a" @ <PATH_TO_X>/test_for.x:6:53-6:54: INVALID_ARGUMENT: Provided status is not in recognized error form: NOT_FOUND: Could not find bindings entry for identifier: "a" @ <PATH_TO_X>/test_for.x:6:53-6:54
F0422 08:51:22.003547    6664 interpreter_main.cc:147] Check failed: ::absl::OkStatus() == (status) (OK vs. NOT_FOUND: Could not find bindings entry for identifier: "a" @ <PATH_TO_X>/test_for.x:6:53-6:54)
```

If you specify a constant, it goes fine.
Please, is there any workaround to use the variable?

cdleary, 2021-04-22:
Hi @pavel532 -- the answer here also relates to https://github.com/google/xls/issues/401#issuecomment-824609148 -- since by default we do pipeline generation mode at the moment (and need to expose the support for sequential generation via a handy tool) we need upper bounds on loops to be compile-time constants so they can be unrolled in space. With FSMs they "roll up" on themselves in time and reuse the same functional units over a bunch of time steps in the resulting RTL, which is why it's a different mode that can support things like `while` loop, by advancing a "virtual program counter" through the original sequence that can spin for an "unbounded" amount of time.

Hope that makes sense, let me know if it clarifies. Am going to take an action here to see if we can expose the sequential generator via a tool (just like the pipeline generator codegen currently is).

cdleary, 2022-11-29:
Note that proc support now lets you iterate in time easily at the user level: https://google.github.io/xls/tutorials/intro_to_procs/

Having a for syntax that desugars into "iterating in time" is what I'd leave this open for -- generally we've been trying to make time more explicit in the XLS programming model as a way to more easily understand the synthesis outcomes, so this is likely a "long term enhancement" in the sense it is potentially nice syntactic sugar (a productivity-oriented iterate-in-time syntax) on top of existing facilities (iterating in time via procs).