google/xls #408: `dslx_jit_wrapper` needs to anonymous namespace ir_text to avoid ODR violations?

cdleary, 2021-04-22:
Noticed the source files all have `ir_text[]` in the xls namespace -- even though it's `constexpr char char ir_text[]` still can cause ODR violations I presume (and these I imagine would be confusing since we'd JIT the wrong thing)?

meheff, 2022-10-04:
Related issue: https://github.com/google/xls/issues/733

I'm guessing this issue can be fixed by putting ir_text into an anonymous namespace.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): All JIT wrapper templates (`xls/jit/jit_function_wrapper_cc.tmpl:28`, `xls/jit/jit_proc_wrapper_cc.tmpl:23-52`, and `xls/jit/jit_block_wrapper_cc.tmpl:25`) now wrap `kIrText`, `kFunctionName`, and `kAotEntrypointsProtoBin` as `static constexpr` inside an anonymous `namespace { ... }` in the generated `.cc` implementation file. (Closed by Gemini)

closed 2026-11-14 by allight