google/xls #411: Map channels to nodes in package

meheff, 2021-04-27:
Currently there is no mapping from channel to nodes in an IR package. This makes it awkward, for exmaple, to find the send/receive node assocated with a channel. We should add such a map. The downside is the necessity of keeping the map up to date as send/receive nodes are added/removed from the graph.

meheff, 2022-03-15:
We do have a similar mapping for register -> registerWrite/Read nodes. This is maintained by code in Block::AddRegister and BlockRemove and overrides of AddNode and RemoveNode. However the channel map is a bit more complicated. Channels are package scoped, but the nodes are functionbase-scoped. This means that the mapping may be modified when nodes are added/removed from the graph AND when functionbases are added/removed from the package. The latter occurs, for example, in FunctionBuilder::Build.

meheff, 2022-03-15:
@hongted

cdleary, 2022-11-01:
Marking as "long term enhancement" because I believe there's not a pressing need to have this association today.