google/xls #416: JIT wrapper/creator should log error & fail for zero-arg functions

RobSpringer, 2021-04-27:
Trying to create a JIT wrapper for the function
```
 fn a() -> u32 {
  u32: 1
}
```

will result in a header like
```
class A {
 public:
  static absl::StatusOr<std::unique_ptr<A>> Create();
  IrJit* jit() { return jit_.get(); }

  absl::StatusOr<Value> Run();
  absl::Status Run(PackedBitsView<32> result);
  absl::StatusOr<uint32_t> Run();
  ...
```

This will fail to compile, since the 1st and 3rd overloads don't have args to differentiate them (no original args = no Run() args). In practice, JIT-compiling a zero-arg function is a bit silly - it's just a fancy way of accessing a constant.

I think a reasonable thing to do, given that, is just to declare zero-arg functions as unjittable and to report a useful log message and abort should one be specified.

cdleary, 2021-04-28:
I think since it's a supported construct in the language, we probably don't want to error when you try to JIT convert it (on the basis of "ideally all language constructs can be JIT'ted) -- can we make `RunValues` distinct from `RunPacked` and `Run`, given `Run` is the most "C++ natural" one.

cdleary, 2021-06-09:
It'd also be nice if Run() returned a value instead of doing an outparam. Will file a bug for that (related) consideration of the JIT wrapper API.

cdleary, 2021-06-09:
Hmm sorry, I guess that's not possible since the idea is the caller allocated it on the stack to be filled in. Duhr, sorry was having a moment. :-) I guess it'd be nice if we somehow noted only the outparam was mutable so we get some const correctness in C++ land. I think I can file for that.