google/xls #418: Unsupported character: '_' Error in Netlist Interpreter (may be related to VAST)

vincent-mirian-google, 2021-05-03:
When using the verilog file:
module __a__a(
  input wire clk,
  output wire [31:0] out
);
  // ===== Pipe stage 0:
  wire [31:0] p0_literal_1_comb;
  assign p0_literal_1_comb = 32'h0000_0001;

  // Registers for pipe stage 0:
  reg [31:0] p0_literal_1;
  always_ff @ (posedge clk) begin
    p0_literal_1 <= p0_literal_1_comb;
  end
  assign out = p0_literal_1;
endmodule

I receive the following error.

F0502 19:43:27.783431 1020326 netlist_interpreter_main.cc:170] Check failed: ::absl::OkStatus() == (xls::RealMain(netlist_path, cell_library_path, cell_library_proto_path, module_name, inputs, output_type, dump_cells)) (OK vs. UNIMPLEMENTED: Unsupported character: '_' (0x5f) @ 1:8) 

It is referring to the '_' in the module name: __a__a in the verilog file.

The commandline is:
bazel run //xls/tools:netlist_interpreter_main -- --cell_library=$PWD/cells.lib    --netlist=$PWD/a.v --module_name=__a__a --input="high"

The contents of cells.lib does not seem relevant to the error.

On a side note:
The verilog was generated using the IR converter, IR Optimizer and Codegen tools.
The DSLX function is:
fn a() -> u32 {
  u32: 1
}

Regardless, the verilog seems correct.

cdleary, 2021-05-21:
I think this has the same root cause as https://github.com/google/xls/issues/419#issuecomment-832930648 (the netlist interpreter is not usable for behavioral Verilog modules).

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `Scanner::PeekInternal()` in `xls/netlist/netlist_parser.cc:288` now accepts leading underscores (`if (isalpha(c) || c == '\\' || c == '_') return ScanName(...)`), and `xls/netlist/netlist_parser_test.cc:92` explicitly tests netlist wires with leading underscores (`_a`). (Closed by Gemini)

closed 2026-11-14 by allight