google/xls #424: Formatted trace (or log)

cg31, 2021-05-05:
The document says "trace! is printf-style", but it seems it doesn't work like that.

Could it be possible to improve it (or add a log!) working like c-printf or rust println!, so we can trace (or log) like:

trace!("[module1] clock {}", clock);

cdleary, 2021-05-05:
Thanks for filing! This is in the works (via @nanavati ) -- can you check out https://github.com/google/xls/issues/403 and note if there are additional aspects you're interested in beyond what's noted there?

nanavati, 2021-05-05:
Here's a couple of specific questions that you could consider commenting on in #403 

1. Are you looking for support just in the DSLX interpreter or also during JIT execution of programs and/or Verilog simulation of generated RTL?
2. If you want trace support in multiple contexts, what your relative priority among the different options?

cg31, 2021-05-06:
duplicate, I will comment in #403

closed 2021-05-06 by cg31