google/xls #425: The file ir_converter_main.py is gone

KuiWei004, 2021-05-06:
Hi all,

When i run the following cmd `$ ./bazel-bin/xls/dslx/ir_converter_main /tmp/simple_add.x > /tmp/simple_add.ir`, i got this error:
`AssertionError: Cannot exec() '/home/xls-developer/xls/bazel-bin/xls/dslx/ir_converter_main.runfiles/com_google_xls/xls/dslx/ir_converter_main.py': file not found.`

It seems that this py file has been removed from xls.

cdleary, 2021-05-07:
Hi @KuiWei004 -- that's right, the Python binaries for the frontend all changed to be C++ a while back, if you pulled a new version of XLS please rebuild via `bazel build -c opt //xls/dslx:ir_converter_main` -- then you should be able to run `./bazel-bin/xls/dslx/ir_converter_main --help` again, let us know if not!

cdleary, 2021-05-25:
Closing this out since I think that'll fix your issue, please reach out / reopen if that didn't solve your issue!

closed 2021-05-25 by cdleary