google/xls #429: Documentation issues

pavel532, 2021-05-13:
On the  https://google.github.io/xls/optimizations/ page 
formatting is displayed incorrectly, it is difficult to read what is meant. 

Perhaps there are the same errors on other subpages https://google.github.io, I have not checked.
[image]

cdleary, 2021-05-13:
Thanks for filing! This happened because there's a slightly different rendering of Markdown we use "in house" from the one supported by mkdocs -- I'll try to comb through and see if there are more blaring ones but let us know if you spot more categories (aside from these style/MathTex ones which are now on my radar).

cdleary, 2021-05-13:
(Also note for that screenshot in the optimization we're likely going to replace it with flowing a little BDD around inside the logic regions, cc @meheff not sure if we have a bug filed for that.)

cdleary, 2021-05-13:
Added support for MathJax so at least the equations should render more nicely now. Still need to solve for image styling.

cdleary, 2023-08-10:
I don't see anything in a search for `margin:` or `style=` in the mkdocs-generated search so I think this may generally be fixed, and the MathJax plugin is active. Please open a new bug for any other weird quirks in the documentation you see!