google/xls #430: exposing packages as public

cg31, 2021-05-14:
I am trying to use XLS as a library.

But it seems I can only put my project, which has cpp file and needs to include and link to XLS, inside XLS tree. Because it looks like all XLS packages are still declared "//xls:xls_internal".

xls/BUILD says they will be exposed as public eventually, when will that happen?

cdleary, 2021-05-17:
Hi @cg31 we started making a public/ directory to expose more stables APIs recently here: 

https://github.com/google/xls/tree/main/xls/public

Can you tell me more about what you'd like to see well exposed in the public APIs?

allight, 2026-11-14:
Keeping this issue open (reviewed by @allight): While XLS exposes headers under `xls/public/` (`xls/public/ir.h`, `xls/public/function_builder.h`, `xls/public/value.h`, `xls/public/runtime_build_actions.h`, and `xls/public/c_api.h`), this is not really what the original issue was asking for regarding package visibility, so we are not closing this issue. (Summarized by Gemini)