google/xls #436: Support Verilog `cover property` statements

RobSpringer, 2021-05-21:
Users have identified this as a feature of real benefit, so it's time to plumb it through. A lot of this will be similar in implementation to asserts, so hopefully it won't be too bad.

cdleary, 2021-05-25:
Just a side note that I thought of as I applied more labels, I think the plan for now is also to support them code generating, but not the ability to get coverage information out of the JIT / interpreter, at least at first, but that'll be nice to have for coverage of JIT based testbenches.

RobSpringer, 2021-05-25:
My thoughts exactly - plumb it through to Verilog, and leave the higher-level stuff as a follow-on.

cdleary, 2021-06-17:
I don't think iverilog has support for cover points, so I think maybe this feature being done is hard to demonstrate with our current open source RTL simulator strategy. cc @mithro in case he has any info on coverpoint support in the open RTL sim ecosystem?

RobSpringer, 2022-04-22:
I think nearly a year is long enough to have this gestate. Closing.

closed 2022-04-22 by RobSpringer