google/xls #442: Missing some DSLX/IR related information in Document

zihaoli-cn, 2021-05-31:
- Several keywords in DSLX's scanner, including "carry", "next", "while", and "proc", don't appear in the document(https://google.github.io/xls/dslx_reference/). It would be better to provide some examples about the usage of these keywords in the DSLX program. 

- The document of Chisel's IR, FIRRTL, not only provides a well-defined semantics specification but also provides a lot of examples to illustrate the usage. However, the concept of "channel" and "proc" in IR, are hard to understand without examples.

RobSpringer, 2021-06-01:
Hey Zihao! That's a good call. Some (all?) of those keywords are intended for features that will arrive Real Soon Now, but we should at least mark them as reserved until they land. I'll add that soon. And yeah - once procs/channels land, we'll need a bunch of documentation to demonstrate their usage. Really, like most projects, we could improve our documentation in a lot of places...we'll chip at it one thing at a time...

cdleary, 2021-06-01:
I think we can turn the google doc for XLS procs and channels in the IR into markdown at this point, cc @meheff for thoughts on that. You can only drive creation of procs/channels at the IR level currently, vs straight from the DSL, but we expect we'll be working on that shortly.

cdleary, 2021-06-01:
I'm going to retitle the issue just because the DSL support doesn't exist yet, but the IR one does, so this issue I thin makes sense to track landing that. Thanks for filing Zihao li!

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `docs_src/ir_semantics.md:84-110, 1294+` documents IR procs, state elements, channels, `send`, and `receive` with examples, and `docs_src/dslx_reference.md:2391+` (`Communicating Sequential Processes (AKA procs)`) documents DSLX procs and channels. (Closed by Gemini)

closed 2026-11-14 by allight