google/xls #445: Support parameterized types in the DSLX-to-C++ transpiler

RobSpringer, 2021-06-09:
(Note that I'm currently working on said transpiler)

On the first pass, we won't support instantiating parameterized types in the transpiler, because A) my use cases don't need it, and B) its hard, but it would be good to have. It'd require some thought as to what we'd be targeting and how we'd want to expose this to a user - are we targeting the JIT? Then we _could_ emit fully instantiated parametrics. Would we be targeting arbitrary usages? Then we'd want...templatized C++ types? Or regular C++ types that could be size-adjusted dynamically?

Not a trivial task.

cdleary, 2022-11-02:
Note the transpiler is just for types, not for functions/expressions. It lives in https://github.com/google/xls/blob/main/xls/dslx/cpp_transpiler_main.cc -- maybe we should move it to the `tools/` directory for consistency/discoverability.

lpawelcz, 2023-11-07:
@proppy This would be super useful in our work on Zstandard codec.

In https://github.com/google/xls/pull/1167, https://github.com/google/xls/pull/1168 and in future work we want to provide C++ tests that process data with `libzstd` and compare the results against simulation of our zstd blocks.
However we are concerned about readability of the syntax used for specifying data for our DSLX structures. For example for [FrameHeaderResult](https://github.com/google/xls/pull/1168/files#diff-9fd280f4a8ee5276a4c77b4d328de166f4ead8bad93af0fb4fb74fda385c0798R56-R61) structure it looks like this:
```
      Value::Tuple(
          {/*status:*/ Value(UBits(0, 2)),
           /*header:*/
           Value::Tuple(
               {/*window_size:*/ Value(UBits(0x1234567890ABCDEF, 64)),
                /*frame_content_size:*/ Value(UBits(0x1234567890ABCDEF, 64)),
                /*dictionary_id:*/ Value(UBits(0xCAFE, 32)),
                /*descriptor:*/
                Value::Tuple({/*frame_content_size_flag: */ Value(UBits(3, 2)),
                              /*single_segment_flag: */ Value(UBits(1, 1)),
                              /*unused: */ Value(UBits(0, 1)),
                              /*reserved: */ Value(UBits(0, 1)),
                              /*content_checksum_flag: */ Value(UBits(0, 1)),
                              /*dictionary_id_flag: */ Value(UBits(2, 2))})}),
           /* buffer:*/
           Value::Tuple({/*contents:*/ Value(UBits(0, 128)),
                         /*length:*/ Value(UBits(0, 32))})})
```
It is not that hard to make a mistake and provide incorrect data. Because of that we wanted to write C++ structs that will map into our DSLX structures and provide a cleaner and less error-prone way of specifying test data. Then we discovered [cpp_transpiler](https://github.com/google/xls/blob/main/xls/dslx/cpp_transpiler/cpp_transpiler_main.cc) which does this for us by generating struct definitions, `ToValue()` converters and `ToString()` methods which are really useful for debugging. Sadly, we encountered some blockers.

One of them is lack of support for parameterized types. For example, our [buffer](https://github.com/google/xls/pull/1167/files#diff-2b3dd20a606f808294f10a32e0dfb91ffcbd6eef0dda2d0270915de501a513c3R27-R30) type cannot be used for generating C++ structs because of `BSIZE` parameter.
We would greatly appreciate having this feature, please let us know what do you think about that.