google/xls #449: ir_converter should be able to convert "many" .x files

cdleary, 2021-06-24:
Right now the IR converter wants to know what the entry module (and, via a command line flag, entry function within that module is) is so it can just convert that one entry module and what it transitively imports.

I think #447 reflects that if we have many `srcs` (.x files) in a "DSL library", we don't know which would be the entry (they could be ordered e.g. alphabetically, there doesn't have to be a "main" one). So it would be useful to convert all of the DSL files in that `srcs` list to a single IR package, where any of those modules / any public function in those modules could be used as an entry point for the downstream tools (e.g. `opt_main`, `codegen_main`).

Thankfully because we mangle the names I think we'll be able to do that (just make a single IR package that encapsulates the symbols from many different .x files).

I think the preference / user guideline is still to try to keep one `srcs = ["foo.x"]` per library wherever we can (generally we try to do that in C++ as well), but I think supporting multiple srcs in the list makes the DSL rules look more like other things in the ecosystem.

cc @vincent-mirian-google and @meheff to check I've got that thinking right / have noted all the salient points -- I think this is part of a broader question of "how do we want our linkage model to work (i.e. in the long run) and be reflected in our build system".

cdleary, 2021-06-24:
Aside: sorry, I realize I'm implicitly also using ".x file" and "module" interchangeably in the above, current terminology is that a `.x` file is a DSL module (can be imported, but solely in a DAG, like in Go). I checked and looks like Go supports multiple `srcs` as well in a `go_library`, though I didn't find anything on the motivation for that decision in the time I had for that bit of background research.

meheff, 2021-06-25:
That sounds right to me. For now our "linkage model" can be throw all the IR in the same file which thankfully should work because of name mangling.

vincent-mirian-google, 2021-06-25:
Right. That is correct. (Tools such as the DSLX converter takes a single input-file as a source.)

cdleary, 2021-06-25:
Great thanks! I was having trouble understanding why taking multiple sources as input would help, but now I think I've connected the dots in my mind successfully. :-) I think this might not be hard to tackle modulo unknown unknowns.

closed 2021-07-28 by copybara-service[bot]