google/xls #470: Clarify fail!() semantics/behavior

RobSpringer, 2021-07-30:
We recently had some confusion over the behavior of the fail() predicate in XLS.

Consider the pseudo DSLX:
```
fn foo(a: u32) -> bool {
  let x = bar(a);  <-- Will fail!() on a == alpha
  let y = baz(a);  <-- Will never fail!()
  y if a == alpha else x
}
```

A user expected that this code would never fail, whereas our targeted use case was that it would not. Our understanding is that the sort-of matching pseudoSystemVerilog could also fail:
```
module foo
input a;
output o;
X = bar(a); <= asserts on input alpha
Y = baz(a); <= does not assert on alpha
o = Y if a == alpha else X
endmodule
```

I'll double-check that SV's behavior matches our expectations, and if so, I'll update our docs to make that clear, I suppose.
If it does NOT match our expectations, then...ick.

RobSpringer, 2021-08-02:
Chris (Leary) and I were chatting about this, and we think it's worth investigating what it'd take to implement semantics like the user's expectation; it could actually be useful for automatically propagating error statuses (into a top-level error vector). I'll write up a one-pager (internally, I think) to flesh this out.

RobSpringer, 2021-08-02:
We don't really have a place for one-pager-style docs to live in OSS, maybe. IDK; I'll figure it out.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): `fail!` and `assert!` semantics are now documented in `docs_src/dslx_reference.md` and lowered to IR `Op::kAssert` tokens via `CallingConvention::kImplicitToken` in `xls/dslx/ir_convert/function_converter.cc`. (Closed by Gemini)

closed 2026-11-14 by allight