google/xls #477: Sanitize verilog keywords

meheff, 2021-08-09:
Currently, codegen will create identifiers in Verilog which are Verilog keywords. We should sanitize those, perhaps by adding a trailing underscore. Common problematic keywords here are "input" and "output"

meheff, 2021-08-18:
An additional problem is where an interface-visible name (i.e., port name) is given a invalid identifier such as a verilog keyword. In this case we should return an error as the expectation is that names of top-level arguments (which becomes ports) should not be changed.

hongted, 2021-11-10:
For interface-visible names, discovering late in the process that an name needs to be changed could have more widespead impact -- especially if modules and tests were already written before the final codegen.  Also, depending on the target language -- verilog, system verilog, VHDL, (whatever next-generation HDL language is adopted).

Suggest that as part of codegen configuration to support user-defined transform/sanitation of port names.

As an example -- for streaming channel inputs allow the user to define the suffix of data/ready/valid.

closed 2024-06-14 by copybara-service[bot]