google/xls #479: Change naming convention of pipeline combinational signals

meheff, 2021-08-11:
We prefix names of signals in pipeline stage with pX_ where X is the pipeline stage. Registers and the combinational logic feeding these registers are prefixed similarly. For example:

```
// pipeline stage 0
...
reg p0_gug;
reg p0_qux;
always_ff ...

// pipeline stage 1
wire p1_foo;
wire p1_bar
assign p1_foo = p0_gug || ...;
assign p1_bar = p0_qux + 1;

reg p1_baz;
always_ff @ (posedge clk) begin
  p1_baz <= p1_foo + p1_bar + p0_gug;
end

// pipeline stage 2
wire p2_bbb;
assign p2_bbb = p1_baz + ...;
...
```

A problem with this naming scheme is that you end up with logic which contains mixed prefixes (e.g., "p0_" and "p1_" in "p1_baz" assignment). Ideally the prefixes should help identify incorrect mixed stage logic.

A better scheme would be to prefix similarly the registers and the combinational logic which use the register values. The above example becomes the code example below. Note the rearranging of logic relative to the pipeline comment delimiters too. Essentially a stage is redefined to be pipeline registers and the uses of those registers, rather than the pipeline register and the logic feeding those registers.

```
// pipeline stage 0
reg p0_gug;
reg p0_qux;
always_ff ....

wire p0_foo;
wire p0_bar
assign p0_foo = p0_gug || ...;
assign p0_bar = p0_qux + 1;

// pipeline stage 1
reg p1_baz;
always_ff @ (posedge clk) begin
  p1_baz <= p0_foo + p0_bar + p0_gug;
end

wire p1_bbb;
assign p1_bbb = p1_baz + ...;
...

// pipeline stage 2
```

RobSpringer, 2021-08-11:
Just note that this might require changes to the per-stage LEC code. I'm willing to handle it, but it might be worth taking a few minutes to think about how we might lighten the coupling between the two here - really, I think just "publishing" the pattern somewhere (e.g., some format details in a header?) would be 90% of the work.

meheff, 2021-08-11:
Does the LEC code key off the name prefixes?

Since we now have registers represented in IR. I wonder if we should do some kind of data flow on the IR and pick out the reg->logic->reg paths from the IR and compare against the net list as the register names will be stable.