google/xls #480: Replace if-else dynamic_cast chains for ConcreteType with a proper visitor

nanavati, 2021-08-11:
See, as one of several examples, [the TypeToIr method](https://github.com/google/xls/blob/b6a4ec7190049002be1c0829d52b1d7767c88204/xls/dslx/ir_converter.cc#L2782) used to convert frontend ConcreteTypes into IR types during IR conversion.

nanavati, 2021-08-11:
But we'd want a visitor before adding new functionality requiring similar traversals (like type-driven "smart" traces).

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): `ConcreteType` (now `Type` in `xls/dslx/type_system/type.h:142,234`) defines `TypeVisitor` and `TypeVisitorWithDefault` with `virtual absl::Status Accept(TypeVisitor& v) const = 0`, replacing `dynamic_cast` chains in `TypeToIr` (`xls/dslx/ir_convert/ir_conversion_utils.cc:45-105`), `xls/dslx/type_system/type_info_to_proto.cc`, `xls/dslx/type_system/type_zero_value.cc`, `xls/dslx/type_system/zip_types.cc`, and `xls/dslx/type_system/deduce_utils.cc`. (Closed by Gemini)

closed 2026-11-14 by allight