google/xls #497: Support declaring arrays-of-channels

RobSpringer, 2021-09-16:
(...and arrays of arrays of channels).

Currently, to create a DSLX design for a 4x4 matmul (commit coming shortly), one needs to declare *FORTY* channels. this is rather unwieldy. If we could instead declare arrays of channels, that'd greatly simplify specifying such a design.

Do it.

lachlansneff, 2021-12-09:
How are channels represented internally? They're not exactly values I would think, since they represent hard-wiring.

RobSpringer, 2022-09-06:
Done a long time ago.

Regarding representation: it varies at the level of interpretation: in DSLX, they're InterpValues, so they can, in some sense, be handled like other values. Inside, they're std::deque<InterpValue> (though issue #664 describes some handling to be improved). In the IR, they're top-level constructs that are handled along the lines I think you're describing.

closed 2022-09-06 by RobSpringer

proppy, 2024-05-30:
Related #704