google/xls #504: Figure out procs & coverpoints/assertions

RobSpringer, 2021-10-01:
I've avoided thinking about how coverpoints & asserts should work inside procs, instead focusing on basic functionality, but as those first bits land, it's time to start thinking about how we thread tokens into and out of procs. It might be as simple as...threading the input tokens through (and restricting covers/asserts to Proc "next" blocks), but regardless, we need to think about it.

taktoa, 2023-02-21:
I ran into this recently, for what it's worth.

allight, 2026-11-14:
Keeping this issue open (reviewed by @allight): While DSLX procs now lower `assert!`, `assert_eq!`, `fail!`, and `cover!` to IR in `xls/dslx/ir_convert/function_converter.cc:2515-2680` and `xls/codegen/block_conversion.cc`, `cover!` semantics in procs remain ambiguous, so we are not closing this issue. (Summarized by Gemini)