google/xls #526: Should DSLX be even Rustier?

lachlansneff, 2021-11-30:
It seems like dslx is already similar to Rust at a surface level, but there are cases that are different for no apparent reason.

For example, why are arrays `T[N]` instead `[T; N]`, like in Rust?

Should types have methods like in Rust? It would certainly make a lot of code cleaner.

```rust
struct F32 { ... }

impl F32 {
    fn add(self, other: F32) -> F32 { ... }
    // ...
}
```

What about traits and operator overloading?

cdleary, 2022-05-10:
@RobSpringer was looking into whether there was an effective way to switch to the Rust multidimensional array syntax.

Goal is to mimic Rust as much as possible, so as you cite structs should have methods and we should have monomorphized traits (no dynamic traits given "synthesizable, familiar, and not surprising" is the DSL's intended constraint). Largely in terms of the project proceeding with end to end deliverable results this was a solid MVP in the state it currently was in, and we dove into other areas that "push the envelope" like support for / definition of procs, but I expect we'll get back to progress on this frontier in this quarter. I'll break out individual issues for the individual topics you cite here.

cdleary, 2022-05-10:
Thanks for raising, filed the issues for tracking, will try to keep the kanban board updated to reflect what we're actively tackling at any given time here for the frontend work.

closed 2022-05-10 by cdleary