google/xls #527: Create checksum-validating options to build rules

RobSpringer, 2021-12-01:
For versioned files, i.e., "released" IPs, it can be VERY useful (ask me how I know) to ensure that the generated contents never change, IOW that we didn't accidentally miss some external, non-versioned dependency.

To do this, we can add a checksum to a Verilog-generating rule and modify said rule such that it compares the checksum of a newly-generated file with that specified in the rule, and, if they mismatch, flag an error.

vincent-mirian-google, 2021-12-01:
Rob. I thought of a solution where we can append a test to all targets via a macro that get triggered on an optional argument (e.g. the golden md5sum - "md5sum_golden = DEADBEEF") . To implement this solution, I would recommend some slight restructuring. I am open to help. Cheers.

vincent-mirian-google, 2021-12-04:
Currently, there is xls_verify_checksum as a solution.

Assign to @vincent-mirian-google for an alternative solution.

vincent-mirian-google, 2022-01-20:
This issue is resolved. xls_sha256sum_test and xls_sha256sum_frozen have been added to the rule list.

closed 2022-01-20 by vincent-mirian-google