google/xls #530: Can't build on MacOS

lachlansneff, 2021-12-05:
I'm using MacOS Big Sur 11.6 and bazel 4.2.1.

When I run `bazel build -c opt -- //xls/... -//xls/contrib/xlscc/...` on the latest `main` commit, I get this error:

```
ERROR: /private/var/tmp/_bazel_lachlansneff/d0308d2008c79c822123b02f33196b89/external/com_google_absl/absl/synchronization/BUILD.bazel:68:11: Linking external/com_google_absl/absl/synchronization/libsynchronization.so failed: (Aborted): cc_wrapper.sh failed: error executing command external/local_config_cc/cc_wrapper.sh @bazel-out/darwin-opt/bin/external/com_google_absl/absl/synchronization/libsynchronization.so-2.params

Use --sandbox_debug to see verbose messages from the sandbox
ld: illegal thread local variable reference to regular symbol __ZN4absl12lts_2021032413base_internal19thread_identity_ptrE for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
Error in child process '/usr/bin/xcrun'. 1
external/local_config_cc/cc_wrapper.sh: line 69: 19620 Abort trap: 6           "$(/usr/bin/dirname "$0")"/wrapped_clang "$@"
```

I tried googling this error, and it seems like it could be an issue with an older version of abseil.

RobSpringer, 2021-12-06:
Well, that's a new one for me. It looks like the root cause is inside Bazel
itself, per these <https://github.com/abseil/abseil-cpp/issues/848> threads
<https://github.com/bazelbuild/bazel/issues/4341> and this one. Can I ask
you to see what version/date you're running? I'll check to see if we're
pinned to a bad release, too.

lachlansneff, 2021-12-06:
For sure! I’m using bazel 4.2.1, which I believe is the most recent version.

closed 2022-01-26 by copybara-service[bot]

cdleary, 2022-01-27:
@lachlansneff I updated Abseil, hope that solves the issue, but we don't have a way to verify on OS X machines ourselves, so just trying to see if that helps "best effort" style. If you get a chance to try it out let us know if it helped!

commit 43095aab (2022-01-26): [deps] Upgrade abseil-cpp to latest LTS release.

Hopefully fixes #530 (but we dont' have OS X machines at our disposal so no way
to verify).

PiperOrigin-RevId: 424214144