google/xls #545: Add block support to eval_ir_main

meheff, 2022-01-10:
eval_ir_main only supports functions at the moment. https://github.com/google/xls/issues/534 discusses adding procs. We also need block support.

We have the ([block interpreter](https://github.com/google/xls/blob/main/xls/interpreter/block_interpreter.h)) which provides an API for interpreting blocks. It needs to be tied into eval_ir_main.

The couple big complications with blocks is multi-output and notion of cycles. We'll need some way of specifying particular inputs at particular cycles and expecting/extracting outputs similarly.

meheff, 2022-01-10:
Also, for blocks which implement ready/valid interfaces it'd be useful to have the tool take care of the flow control and the user can just specify the inputs and get the respective data outputs.

meheff, 2022-01-10:
Another nice feature would be to support VCD format. We could both read in VCD files generated from a external verilog test benchmark, and generate them for use in waveform viewers and other tools for example.

cdleary, 2022-11-02:
Just a note that this issue is still up to date. @grotival is also working on improving block interpreter to work with instantiations.

cdleary, 2022-11-02:
Note the related issue #546