google/xls #551: Add support for instantiation to the IR interpreter

meheff, 2022-01-14:
The IR interpreter doesn't yet support the instantiation operation. The instantiation operation is a low-level op which can appear in blocks. It instantiates another XLS block. It is analogous to a Verilog module instantiation.

Initially, it'd be easiest to support instantiation of combinational blocks only to avoid having to worry about register state. The implementation would be in block_interpreter.cc. It could work by simply creating another [BlockInterpreter](https://github.com/google/xls/blob/c330e64365e56439ab9496159aa8664c6cd5eb6a/xls/interpreter/block_interpreter.cc#L25) object within the HandleInstantiation method of BlockInterpeter. This sub interpreter could evaluate the block and return the outputs and the interpretation could keep on trucking.

Stateful instantiation is much trickier. A block might be instantiated multiple times from different points of the code. Each instantiation needs its own register state. So register state needs to be keyed by instantiation stack (similar to call stack in a normal program) and register name.

proppy, 2026-10-28:
I'm not sure if this qualify this as a `good-first-issue`.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `ElaboratedBlockInterpreter` in `xls/interpreter/block_interpreter.cc:437-583` implements interpreter support for `BlockInstantiation`, `FifoInstantiation`, and `DelayLineInstantiation`, tested in `xls/interpreter/block_evaluator_test_base.cc:260-310, 478-580`. (Closed by Gemini)

closed 2026-11-14 by allight