google/xls #552: Predicated nodes

taktoa, 2022-01-15:
Note: This is somewhat speculative; I haven't worked out all the details. Feedback is welcome.

In theory, there are three situations in which nodes that are implemented by the same hardware (e.g.: two adders of the same bitwidth) may be unified to save area:

1. In pipelines with `II > 1`, if one node is in stage `i` and the other is stage `j`, they may be shared iff `(i - j) mod II ≠ 0`.
2. Clock gated nodes may be shared if the gating predicates are mutually exclusive, i.e.: at most one of the predicates can ever be true at a given time.
3. A generalization of II > 1 would be that nodes may be guaranteed to be used only when the current clock cycle number satisfies `clock mod p = q`, for user-chosen `p` and `q`.

To unify these scenarios, I propose that we introduce a notion of _node predication_. That is, each node has an optional pointer to another node that is considered to be its predicate. Predicates do not affect the semantics of the IR with respect to interpretation, they are merely hints for code generation.

All three of the above scenarios can be handled with this mechanism, and using a node itself as the predicate has the advantage that the predicate nodes are subject to things like common subexpression elimination and dead code elimination, decreasing the memory overhead of predicates.

cc @meheffernan

taktoa, 2022-01-15:
In terms of actually exploiting these predicates, I think you could probably just ask Z3 whether a given pair of predicates is mutually exclusive and get an answer quickly and reliably. However, running Z3 a number of times that is quadratic in the number of nodes is probably a bad idea. You can massively prune the number of Z3 calls by only checking for mutual exclusivity of predicates that share the same variables, since if the variable sets contain non-intersecting elements then there are additional degrees of freedom which could be used to make both predicates true at the same time. Unfortunately this is ruined a bit by tautological variable uses (e.g. P or !P shouldn't really count as a use of P), so I'm still trying to figure out a reliable way to do this.

taktoa, 2022-01-18:
Okay, after thinking about this some more, I have what I think is a fairly reasonable design of a resource sharing algorithm based on these node predicates.

A general overview of the algorithm is that it will consist of the following stages:

1. Use an SMT solver on "all" pairs of predicates to determine when they are mutually exclusive (as pairs). Create a graph `G` with a node per predicated node and edges between any two predicated nodes whose predicates were determined to be mutually exclusive and which are shareable (i.e.: implemented by similar hardware resources). We also get rid of any neighborless nodes, since these were identified to have no sharing opportunities. We don't actually need to test all pairs with the SMT solver -- there are good and quick ways to prune the set of pairs drastically: for example, the set of monotonically used variables must be equal for two predicates to be mutually exclusive.
2. Enumerate all maximal cliques in `G`. This gives us the full clique structure of the graph, since every clique is a subgraph of a maximal one. There are good output-sensitive algorithms for this. The clique structure can be thought of as a hypergraph.
3. Greedily remove the largest edge of the hypergraph, deleting its nodes from all other edges (but keeping those edges around as smaller edges), until there are no edges of size 2 or greater left.

Step 3 works well unless there are significant gains to be made through operator fission. I’ve figured out an optimization problem to solve that fully accounts for operator fission, but I don’t have a good way to solve it yet. The optimization problem is as follows:

Given a hypergraph `H = (V, E)`, we refer to a hypergraph `H' = (V, E')` as a _refinement_ of `H` iff there exists a partition `p : E' → I` such that `E = {∪(p⁻¹(i)) | i ∈ I}` (i.e.: there's a way of merging the edges in `E'` to get the edges in `E`).

Given a hypergraph `H = (V, E)` with node weights `w : V → ℕ`, find a refinement `H’ = (V, E’)` of `H` and a mapping `c : E’ → ℕ` such that if `IncidentSum(v) = sum({c(e) | e ∈ E’, v ∈ e})`, then `IncidentSum(v) ≤ w(v)` for all `v ∈ V` and `sum(map(f, E’)) + sum(map(g, {v ∈ V | IncidentSum(v) < w(v)}))` is minimized, where `f : ℙ(V) → ℕ` and `g : V → ℕ` describe how much area is used by a shared set of nodes (including control circuitry overhead) and how much area is used by a "leftover node" that wasn't shared.

meheff, 2022-01-20:
For predicate which are determined by pipeline stage what would the predicate nodes be? Would these be new nodes added to the graph where there logic is such that collectively the set of predicates is one-hot?

taktoa, 2022-01-20:
The predicate in that case would be `clock % II = (pipeline stage #) % II` where `clock` is a new magic param or something. I haven't figured out the subtleties around that `clock` param, in particular the semantics of its bitwidth. Not sure what you mean by the latter, some of the predicates will be mutually exclusive and others won't be, and yes the predicate nodes are added to the graph.

taktoa, 2022-02-01:
I realized last night that solving the complicated optimization problem described above is unnecessary, at least in the case of sharing adders and multipliers. If you have a clique consisting of adds with varying bitwidth, you can implement all of these using a single adder the width of which is equal to the widest add in the clique by simply zero-extending the addends and then bit-slicing the result. If we adopt this approach, then we simply need to use a min-clique-cover solver to get the best performance out of predicated node merging.

The only issue with this is that you might argue that it would make our delay estimates less accurate, but this pass happens after scheduling, so that should be fine. A bigger issue would be if this makes the delay estimates used by DC less accurate (since they'd need to be conditional on how much of the adder result is actually used in a given codepath), in which case it might do bad optimizations.

taktoa, 2022-02-17:
I've figured out that min-clique-cover can be reduced to graph coloring on the complement of the mutual exclusion graph. Another insight I had is that once we determine a pair of predicates to be mutually exclusive or not mutually exclusive, that information is not invalidated except by `ReplaceOperand` and `AddOperand`, so it makes sense to keep the mutual exclusion information in a field in `FunctionBase` so that it can be used in other passes.

I've split the work for this up into three main features:

1. Computing simple node predicates and determining when they are mutually exclusive
    1. Add metadata for how nodes are predicated by other nodes.
    2. Add metadata for how predicates are mutually exclusive with one another.
    3. Write a pass that propagates `gate` predicates to all postdominated nodes.
    4. Write a pass that propagates `select` and `one_hot_select` predicates to all postdominated nodes.
    5. Write a pass that eliminates predicates from the graph (run prior to codegen).
    6. Write a pass that updates the mutual exclusion metadata based on running Z3 on any pairs of nodes for which mutual exclusion / non-exclusion is not established.
2. Calculating a min-clique-cover and merging nodes based on it
    1. Implement a simple graph coloring algorithm like [DSatur](https://en.wikipedia.org/wiki/DSatur).
    2. Write a wrapper around that that does min-clique-cover on subgraphs of the mutual exclusion graph (stratified by node type).
    3. Write a pass that merges nodes based on the cliques. It will need to generate zero extends and bitslices for arithmetic ops that have cliques of varying bitwidth, and selects that uses the predicates to choose which inputs are given to a merged operation.
    4. Bonus: write a min-clique-cover algorithm using Z3 or enumeration that is guaranteed to attain the optimal value (albeit in worst case exponential time) so that we can benchmark how bad the `DSatur` based algorithm is on our graphs. If it's bad, then I think there's a strong case that the slightly more complicated [RLF](https://en.wikipedia.org/wiki/Recursive_largest_first_algorithm) algorithm will work well.
3. Support for II > 1
    1. Frontend support for declaring the initiation interval of a function/proc, which will be available as an `int64_t` field on `FunctionBase*`. A very simple version of this could be a compiler flag, but something like a Rust attribute would likely have the best developer experience.
    2. A post-scheduling pass (which is a new notion we'll need for rematerialization too) that uses the initiation interval to add predicates of the form `param("clock_mod_${II}") == literal(stage % II)` to the graph.
    3. Adding something to codegen that wires `clock_mod_k` parameters to state machines that implement a counter with `k` states, and that get reset to `0` when an input is `valid` (which allows the II > 1 semantics to be "I promise to never expect the output of this block to be valid twice or more in a `k` cycle period" rather than "I promise to only expect the output of this block to be valid when `cycles_since_boot mod k == block_latency mod k`"; the former is a superset of the latter).

taktoa, 2022-02-18:
In addition to Ted's need for mutual exclusion information on the predicates of unordered send_if nodes, another way the mutual exclusion information could be used is to determine when a OneHotSel truly has a one-hot selector (this happens when the predicates for each case of the `one_hot_sel` form a clique in the mutual exclusion graph).

hongted, 2022-02-18:
I should also note that for mutual exclusion -- we may not want something to be truly one-hot.

In the case of send predicates, it's fine if none of the predicates are true.  We only need the property that if one is true the others are false.  This isn't true one-hot in that there is always one and at most one that is hot.

meheff, 2022-02-19:
This sounds promising. There is a lot here potentially to implement and some uncertainty about the final reward, so I'd recommend trying to find concrete examples in our benchmark set (or outside examples) to drive the development so you know there will be pay off in the end. On that front, implementing a focused, but more limited implementation to realize those gains in the near term is a good strategy. Then this can be built out and generalized when some initial gains have been captured. Roughly, this is a DFS search rather than a BFS search where the potential wins are out at the leaves. This approach can potentially avoid a lot of wasted work.

I'm a bit skeptical about the necessary IR changes. There is a reason we don't have much metadata hanging off the IR constructs. Keeping this metadata up to date can be a challenge and can add significant complexity. If it's possible to construct the needed information in a just-in-time manner that would be preferable, and certainly a much better place to start. Similarly overloading existing IR constructs (for example, by adding magic parameters and or special implicit uses) may be a tough sell. Keeping the IR as lean as possible with a minimal number of special cases or warts is very important to managing complexity throughout the compiler.

taktoa, 2022-02-19:
Regarding Ted's point about the mutual exclusion meaning "one-hot", that was a mixup on my part; I've been using the NAND definition of mutual exclusion, and briefly forgot that when mentioning applicability to one-hot selectors. I think it is still applicable even with that proviso, since there is likely plenty of utility in inferring "this selector is either one-hot or zero" (for example in range analysis, though I think Mark wants to get rid of one_hot_sel anyway).

The idea of running test vectors through the block to determine non-mutual-exclusion is a very good one. Initially I thought that it might be obviated by the fact that Z3 does bitblasting, but upon further thought I think bitblasting the whole graph at once (and for that matter, JITing it) could yield a large increase in speed.

Regarding @meheff's comment:

1. I didn't mean to imply that features 1, 2, and 3 must be completed in that order or at all. I separated 3 from 1 specifically because it involves frontend and codegen work and does not deliver value until we have II > 1 blocks to run through the compiler. The `gate`, `sel`, and `one_hot_sel` predicates are all very similar to implement, at least in a basic way (e.g.: iterate over postdominated nodes), so I don't really see it as much of a distraction to have support for all of them.
2. There's a lot of different things you allude to in the second paragraph, it'd be nice to hear a more specific account for each thing:
    1. Having the predicates be nodes (which may be, for example, DCE'd at the end of one monolithic pass that does 1, 2, and 3, except the `clock_mod_n` state machine generation).
    2. Having the predicates be nodes specified in the `FunctionBase` which are implicitly used.
    3. Having the mutual exclusion information specified in the `FunctionBase`.
    4. Having the initiation interval specified in the `FunctionBase`.
    5. Having magically-named `Param` nodes for `clock_mod_n`, at least temporarily.

Avoiding 2.i would be difficult as it would require creating a new syntax tree for predicates, when we already have a perfectly good syntax tree for predicates. It would also prevent us from using the large array of tools that we have that work well on nodes, like the JIT that Ted mentioned or the Z3 translator. There are clearly cases where we know from first principles that there is mutual exclusion between predicates, so we should definitely take advantage of that, but I don't think that requires adopting a more restricted language of predicates.

Avoiding 2.ii or 2.iii requires making the pass monolithic (for the reasons mentioned in my last comment), which will mean that the mutual exclusion analysis cannot be shared between a pre-scheduling pass (that for example does `sel`, `one_hot_sel`, and `gate`-driven sharing which may require further optimizations) and a post-scheduling pass (that does II > 1 predicates). Maybe Ted's suggestion of bitblasting the graph to find non-mutually-exclusive nodes will reduce the cost of mutual exclusion analysis to the point where it's unnecessary to worry about this, however. Or maybe the potential QoR improvements from the former will not pan out so it won't be a concern anyway.

Avoiding 2.iv seems fairly untenable to me. I guess I could imagine passing it in as part of the `PassOptions` instead, but it's really a immutable fact about the `FunctionBase`, so why wouldn't it be in the `FunctionBase`? It'd be like saying that the `Type` of a node shouldn't be in the `Node`.

Avoiding 2.v requires a fair bit of work that is not really value-generating (even in a long-term sense). We'd have to create a new IR node that behaves identically to a `Param` with regard to all existing passes and analyses, with all the duplicated logic that implies, just to avoid purely hypothetical name collisions, which could be avoided in other ways anyway. Or we'd have to change the `Param` node type to contain metadata specifying whether it's a `clock_mod_n` node or not, which I find less objectionable, but still not high priority. Alternatively, we could do a completely monolithic pass at the `Block` level, where we have the ability to generate these state machines, but that raises so many implementation questions as to be a nonstarter (for example, where do we get schedule information from? it has already been eliminated into pipeline registers at that point).

I already have a partial implementation that uses the 2.ii + 2.iii strategy (which I think are by far the most reasonable of the 5 to avoid), and it'd be work to switch away from it, so I'll likely keep it around until your review unless it causes me grief. The rest seem like unnecessary detours from the more important work of getting QoR wins and the new II > 1 functionality working.

meheff, 2022-03-03:
Regarding a couple specific items in (2) which I don't think are really a problem:

i. Having predicated nodes be actual nodes in the IR is not really a problem. They can be added, used, then stripped away.

iv. Specifying the initiation interval on the function base is fine. We'll likely need other per-function information or hints which either comes from the frontend or are fed in from the outside via some other path. Specifying a particular number of stages (latency) for a particular proc/function may be desirable and is of the same ilk. Right now we handle this via codegen options on the side, but it may be cleaner to attach that information to functions especially when we get to the point of compiling multiple functions into blocks / verilog modules.

The other items are more problematic in my mind as they are adding special purpose constructs to the IR (implicit uses, magic params) or adding metadata to the IR which is derived from the IR and thus can run into coherency problems. Fortunately, I don't think these are strictly necessary. It seems that there are two roughly kinds of use which I'll call spatial and temporal:

1.  spatial reuse: at most one of the results of two (or more) operations is used in a single evaluation of the data flow graph. The case is easier to handle and it can be performed with a simple graph transformation with the existing IR constructs. Inputs are muxed based on the mutually exclusive predicate and the output fans out to the uses of the two original operations. These opportunities might be handled in a single pass which identifies reuse possibilities then transforms the graph.

2. temporal reuse: two operations are never evaluated in the same cycle (example 1 and 3 in the original issue post above). This occurs in II > 1 situations. This is the more complicated issue because of the need to carry this information forward from scheduling and predicate analysis down into codegen. However, I think these cases can be handled via outlining. Specifically, two operations which are to share are single hardware resource would be replaced with invokes of a function which contains the operation. As there are no other invokes late in the pipeline there should be no confusion as to what such an invoke means. This is a relatively natural representation as we have an IR construct (the called function) which represents the shared resource, and the invokes are stand ins for the use of the resource.

   This outlining could be done before, during or after scheduling. If done before scheduling, the scheduler would have to know how to handle these invokes and schedule them accordingly (e.g, they need to be in mutually exclusive cycles). These invokes would be passed on to codegen which (using the schedule) would be able to add the right logic to mux in the proper operands in the right cycles.

   The use of invokes and functions is convenient in the immediate term for representing this transformation, but if this overloading proves awkward then we could think about adding a special operation for this purpose which look like an invokes but means use of a shared resource.

taktoa, 2022-03-03:
@hongted mentioned in the meeting the possibility of procs that are scheduled in a single cycle and use different resources on diferent cycles mod initiation interval based on their state. For example, a proc that has 1 extra bit of state that alternates each cycle and controls what the proc body does to the rest of the state. This can be handled by first unrolling the proc `initiation_interval` times while maintaining a map of what nodes map to what original nodes in the non-unrolled version. Importantly, when we unroll the proc, the state becomes a param, we don't use the initial state at all. Then run Z3 to determine the mutual exclusion graph of the unrolled version. If all "versions" of a node are mutually exclusive with the corresponding "versions" of another node, then add an edge to the mutual exclusion graph for the non-unrolled version.

closed 2022-07-20 by taktoa